ni/nimi-python #1323: The way Driver Version is queried in print_diagnostic_information will not work if only driver runtime is installed

sbethur, 2020-03-12:
### Description of issue

`driver_version` is queried in `print_diagnostic_information` by looking at the value of `HKLM\SOFTWARE\National Instruments\<DriverName>\CurrentVersion` regkey.

This regkey is not part of runtime installer, but rather part of full installer. We should update the logic to get the version of driver runtime and also update the wording to make it clear that the info is for the driver runtime.

#1205 is a related issue. Both should be fixed together.

marcoskirsch, 2021-12-01:
We could get the version of the installed driver by calling `nipkg.exe` on Windows or `yum` on RedHat.

marcoskirsch, 2021-12-01:
#1206 is also related.