ni/nimi-python #1337: Some methods in nidigital could (should?) use `sites` rep cap

marcoskirsch, 2020-03-20:
As of #1290 we have a `sites` repeated capabilities object in `nidigital`. Two methods were not made to take advantage:

1. `fetch_history_ram_cycle_information`
2. `get_history_ram_sample_count`

I suspect this is because they only allow one site at a time. I don't think that's a good reason as we have functions in other APIs that only allow one channel. We let the driver runtime error if the user passes more than one.

```python
hram_info = my_session.fetch_history_ram_cycle_information(3)
```

should become

```python
hram_info = my_session.sites[3].fetch_history_ram_cycle_information()
```

sbethur, 2020-03-20:
`fetch_history_ram_cycle_information` already uses `pins` repeated capability. So, it cannot use `sites` repeated capability; at least with the way we're currently nesting rep caps. Users typically use `get_history_ram_sample_count` and `fetch_history_ram_cycle_information` together. For the sake of consistency, I decided to not use `sites` rep cap for both.

sbethur, 2020-03-20:
`is_site_enabled` is the other method without `sites` rep-cap, which I'd argue should use the rep cap.

marcoskirsch, 2020-03-25:
One way to implement this is to create "fancy" function that parses the `self._repeated_capability` string and breaks it up into two parameters.

PR #1359 (2020-03-31): Update applicable methods in nidigital to use sites repeated capability
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

Update these methods in nidigital to use sites repeated capability:
 - fetch_history_ram_cycle_information
 - get_history_ram_sample_count
 - is_site_enabled

### List issues fixed by this Pull Requ [... cut]
closed 2020-03-31 by marcoskirsch