ni/nimi-python #1341: Programatic use of nidigital API

rml7rt, 2020-03-20:
@markusroeleke, First off, thanks for using the pre-release nidigital API and giving us feedback before we finalize the API. We want to understand your usecase better and have a few questions:

1. Can you please elaborate on how you're dynamically reconfiguring 6570 during runtime?
    - Are you closing and reopening the session to load a new pinmap or are you calling `reset()`?
    - Does the runtime dynamic reconfiguration also involve changes to the physical test setup like changing the loadboard/DUT, rewiring the connections, etc.?
2. How many pinmaps are being used in the test program and how different are they?
3. What kind of DUTs are being tested?
4. Is the test program for characterization or production test?
5. Is TestStand Semiconductor Module being used in any way?

_Originally posted by @sbethur in https://github.com/ni/nimi-python/issues/1124#issuecomment-597327950_

mroeleke, 2020-03-20:
Hi guys, 
thanks for you amazing work you are doing. We use your API heavily with very good results.
To answer your questions:
>     1. Can you please elaborate on how you're dynamically reconfiguring 6570 during runtime?
>        * Are you closing and reopening the session to load a new pinmap or are you calling `reset()`?
yes we open and close sessions and reload diffrent pin maps during runtime. Usually we do not call a reset in between.
>        * Does the runtime dynamic reconfiguration also involve changes to the physical test setup like changing the loadboard/DUT, rewiring the connections, etc.?
Yes, we also change the physical setup in between, since a pin of an ASIC has different capabilities. It can be used for example as an output or an analog testbus. If I use the pin as a digital io i wozuld connected it to the 6570. I f I use it as an analog testbus i would connect it to an PXI4141. This requires a dynamical reconfiguration of the hole system during test execution.
> 
>     2. How many pinmaps are being used in the test program and how different are they?
If I have a 40 pin device I would use > 40 pin maps. The reason is that I want sometime control a pin individually (continuity, input leakage) and use it sometimes as a port (IC2, I3C, SPI3/4, TDM, ...).
>     3. What kind of DUTs are being tested?
Different mixed signal ASICs with an digital interface, as described above and strong analog capabilities.
>     4. Is the test program for characterization or production test?
We use our setup for debug and char.
>     5. Is TestStand Semiconductor Module being used in any way?
No, we use Python3.x. Since we need to have a reproducible system which can be setup later to exact the same setup as before. What if we get a sample back 4 years after development? We do not need a GUI. We need a version controlled script to execute and a version controlled environment to be setup at or system. Anaconda for example does a great job here.
> _Originally posted by @sbethur in [#1124 (comment)](https://github.com/ni/nimi-python/issues/1124#issuecomment-597327950)_

marcoskirsch, 2020-03-20:
>thanks for you amazing work you are doing. We use your API heavily with very good results.

Thank you for the feedback. It's great to hear.

marcoskirsch, 2020-04-15:
@sbethur along with other colleagues at NI have discussed the use case with the customer and concluded that the application is better served using a single Pin Map and Digital Pattern Editor.

Thank you @mroeleke for using `nidigital`, and being in touch with us to make it as good as we can.

closed 2020-04-15 by marcoskirsch