ni/nimi-python #1352: Request for fast programatic pattern creation/compile

mroeleke, 2020-03-25:
Hi Guys,
currently we use the pattern card to write some tests for our product. In order to do that we read and write a sequence of SPI transfers (e.g. read and write) to and from the DUT. That looks like that:

- write('RegisterA', 0x55)
- regb = read('RegisterB')
- write('RegisterC', regb + 5)

That means I can not have a fixed pattern, since the argument for the write cmd to register C is dependant of the current content of register B. 

To use the PXI6570 I have to

1. translate every SPI transfer into a pattern,
2. compile it, 
3. load it to the card, 
4. do a pattern burst, 
5. read back the values and give it back to the function. 

The problem is now is , that I have to compile every single transfer, that takes about 450 ms. The SPI transfer itself takes about 0.01 ms. That means I am a factor 45.000 slower as I could be. We have usually 100 tests with each 200 transfers. So a complete test would take than 9.000 s which is about 2.5 h and way to much. This but it should happen in a matter of minutes. 

In order to use the card effectively, to compile / load a pattern should not take much more than 1 ms.

Do you guys see a way to strip done the compiling time somehow?

sbethur, 2020-04-06:
Hi @mroeleke,

NI-Digital Pattern Driver has a feature called [Source and Capture Waveforms](http://zone.ni.com/reference/en-XX/help/375145e/digipat6570/source-capture/), which was created for situations like you describe. Please refer to the product documentation for how you to use source and capture waveforms and how it can help with your test program.

We also ship LabVIEW, .NET, and C examples that demonstrate how to use source and capture waveforms. Python examples are still in work, but you can refer to the system tests to get an idea of how to use source and capture waveforms:
https://github.com/ni/nimi-python/blob/1dd126e6e22cbe27a129350cc3d26bf32488df5e/src/nidigital/system_tests/test_system_nidigital.py#L163
https://github.com/ni/nimi-python/blob/1dd126e6e22cbe27a129350cc3d26bf32488df5e/src/nidigital/system_tests/test_system_nidigital.py#L199
https://github.com/ni/nimi-python/blob/1dd126e6e22cbe27a129350cc3d26bf32488df5e/src/nidigital/system_tests/test_system_nidigital.py#L244

If you have more questions, please visit [NI Community Forums](https://forums.ni.com) or follow up on NI's [Support Portal](https://www.ni.com/en-us/support.html).

sbethur, 2020-04-07:
Closing this issue since it does not pertain to nimi-python.

closed 2020-04-07 by sbethur