ni/nimi-python #1361: Publish nidigital 0.5

sbethur, 2020-03-31:
### Description of issue

nidigital API has been finalized. Currently, we are working on tests to verify the API. While that is going on, we should publish the current API on PyPI, so users can start using it and provide feedback.

mroeleke, 2020-04-02:
Hi @sbethur,
when will you publish version 0.5?

sbethur, 2020-04-03:
Hi @mroeleke,
We are targeting to release 0.5 once we fix all issues in [API finalization milestone](https://github.com/ni/nimi-python/milestone/20). Currently, it's looking like it will be at least a week to complete that.

If you need a build before that, I can publish 0.4.1 now.

mroeleke, 2020-04-14:
Hi @sbethur,
yes, can you please release a version 0.4.1 shortly?

closed 2020-04-22 by sbethur

sbethur, 2020-04-22:
@mroeleke, nidigital 0.5 is released.

PR #1422 (2020-04-22): Release nidigital 0.5.0 and 1.2.1 of other modules
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

~- [ ] I've added tests applicable for this pull request~

### What does this Pull Request accomplish?

Release the following:
- nidigital 0.5.0
- nidcpower 1.2.1
- nidmm 1.2.1
- nifgen 1.2.1
- nimodinst 1.2.1
- niscope 1.2.1
- nise 1.2.1
- niswitch 1.2.1
- nitclk 1.2.1

### List issues fixed by this Pull Request below, [... cut]