ni/nimi-python #1368: datetime.timedelta does not provide the necessary resolution for many methods and properties in nidigital

sbethur, 2020-04-03:
### Description of issue

`datetime.timedelta` has a microsecond resolution, which is not sufficient for many method parameters, return values, and properties. e.g. edge placement, vector period, TDR offset etc.

sbethur, 2020-04-03:
@marcoskirsch, I see `datetime.timedelta` being used in other nimi-python modules. Is there a resolution issue in any of those too?

marcoskirsch, 2020-04-03:
No, all the places in which we use `datetime.timedelta` we verified the resolution is sufficient.

texasaggie97, 2020-04-03:
There is already an issue for this: #744 

Though so far it hasn't been a real issue, just a theoretical issue.

sbethur, 2020-04-09:
Implementation notes from Mark on how to add `hightime` as a dependency to `nidigital` installer:

> You would need to touch setup.py.mako, tox-system_tests.ini.mako, and tox.ini and then rebuild.

sbethur, 2020-04-29:
#1438 updated nidigital installer to consume hightime library but the code hasn't been updated to use hightime types yet.

PR #1438 (2020-04-29): Add hightime as a dependency to nidigital installer
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

~- [ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~

~- [ ] I've added tests applicable for this pull request~

### What does this Pull Request accomplish?

Add hightime as a dependency to nidigital installer. `hightime` library will be used to fix #1368.

### List issues fixed by this Pull Request below, if any.

Change contributes toward fixing #1368 

### What testi [... cut]
commit a9600a2e (2020-04-29): Add hightime as a dependency to nidigital installer (#1438)

`hightime` library will be used to fix #1368.
PR #1449 (2020-05-19): Replace usage of `datetime` library types with `hightime` types
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

In all nimi-python modules, change the type of applicable properties and method parameters from `datetime.timedelta` to `hightime.timedelta` and from `datetime.datetime` to `hightime.datetime`.

`hightime`  [... cut]
closed 2020-05-19 by marcoskirsch