ni/nimi-python #1371: Segmentation fault when niswitch is used by a different user

cekins, 2020-04-03:
### Description of issue

The niswitch module was installed by user A and functions normally for scripts executed by user A. When user B logs in and attempts to use the same scripts as A, a segmentation fault occurs at the creation of an niswitch.Session

### System report

`python -c "import niscope; niscope.print_diagnostic_information()"` output (replace `niscope` with appropriate package name)

 ```
OS:
    Version:   #1 SMP Tue Aug 14 21:49:04 UTC 2018
    Bits:      64
    Name:      Linux
Driver:
    Version:   Unknown
    Name:      NI-SWITCH
Module:
    Version:   1.1.5
    Name:      niswitch
Python:
    Version:   2.7.10 (default, Dec  4 2017, 12:24:37) 
[GCC 4.3.2]
    Bits:      64
    Is_Venv:   False
    Installed Packages:
        wsgiref==0.1.2
        python==2.7.13
        faulthandler==3.2
        wheel==0.34.2
        urllib3==1.22
        traitlets==4.3.2
        soupsieve==1.9.5
        six==1.11.0
        singledispatch==3.4.0.3
        setuptools==44.0.0
        scipy==1.2.3
        rtmdmetrics==1.2.0
        rtmdhw==0.0.2
        requests==2.18.4
        pytz==2017.3
        pyserial==3.4
        ptyprocess==0.6.0
        plotly==2.2.2
        pip==20.0.2
        pillow==6.2.1
        pexpect==4.6.0
        numpy==1.16.2
        niswitch==1.1.5
        nbformat==4.4.0
        jupyter-core==4.4.0
        jsonschema==2.6.0
        ipython-genutils==0.2.0
        idna==2.6
        functools32==3.2.3.post2
        enum34==1.1.6
        decorator==4.1.2
        chardet==3.0.4
        certifi==2017.11.5
        beautifulsoup4==4.7.1
        backports.functools-lru-cache==1.6.1
[ekinsc@kfokC75 test_scripts]$ 

 ```

### Steps to reproduce issue

    ``` 
    import niswitch

    with niswitch.Session('Dev1') as session:
	print session.get_relay_position(relay_name='k10')

    ```

    1. Install niswitch as a user
    2. Log in as a different user
    3. Attempt to create an niswitch session as the second user

sbethur, 2020-04-06:
Hi @cekins,

1. Which Linux distribution and version are you using?
2. If you got any error traceback before the seg fault, please post it
3. Make sure you did not use `--user` option when pip installing niswitch
4. Does the second user have administrator privilege? As second user, check if creating the niswitch session as root (sudo) fixes the problem.

marcoskirsch, 2020-04-06:
I very much doubt this is a Python-specific issue. @cekins any chance you can build a simple C executable and test that way?

marcoskirsch, 2020-04-06:
Never mind @cekins, this is a known issue (Internal bug number 963590).

If you give all users read/write access to files `/var/lib/ni-switch/lock` and `/var/lib/ni-switch/shm` then the problem should go away.

Please give it a shot and report back.

cekins, 2020-04-07:
We changed the permissions and it works now.

Thanks for the reply @marcoskirsch

closed 2020-04-07 by cekins