ni/nimi-python #1386: Improve usability of get_channel_name in nidigital API

sbethur, 2020-04-07:
### Description of issue

Currently, there are two very similar methods in the public API:

1. `get_channel_name()`
    - Accepts a single channel number as `int`.
    - Range is [1 : ChannelCount * InstrumentCount]
    - One-based channel input
    - On multi-instruemnt session, returns a string that represents a single fully-qualified channel name e.g. 'PXI1Slot2/0'
    - On single-instruemnt session, returns a string that represents a single channel number e.g. '0'
2. `get_channel_name_from_string()`
    - Accepts a `str` which can be one of the following formats:
        - A comma-separated list - for example, "0,2,3,1"
        - A range using a hyphen - for example, "0-3"
        - A range using a colon - for example, "0:3"
    - Range is [0 : (ChannelCount * InstrumentCount) - 1]
    - Zero-based channel input
    - On multi-instruemnt session, returns a comma-delimited string of fully-qualified channel names e.g. 'PXI1Slot2/0,PXI1Slot5/31'
    - On single-instruemnt session, returns a comma-delimited string of channel numbers e.g. '0,31'

Proposal:
1. Get rid of `get_channel_name()`
2. Rename `get_channel_name_from_string()` to `get_channel_name()`
3. Make the new `get_channel_name()` use `channels` repeated capability - This will make this method consistent with rest of the API w.r.t. using `channels` repeated capability and it will allow the implementation to leverage support for ranges,lists,slices etc. that are available for repeated capabilities.

marcoskirsch, 2020-04-07:
This does not work.
The `get_channel_name_from_string` method takes in 0-based _channel indices_. The repeated capability is for _channel names_, which in multi-instrument session **must be** fully qualified channel names like "Dev1/4". 

We cannot mix the two concepts.

We still need to figure out what to do about these methods though.

sbethur, 2020-04-10:
After discussing offline, the proposed solution is:

Any module in nimi-python that needs to provide an API for translating channel indices to names will have only one method - `get_channel_names(channel_indices)`.
  - `channel_indices` input represents channels numbers. Supported types:
    - str - List (comma delimited)
    - str - Range (using '-' or ':')
    - str - single item
    - int
    - list of str
    - tuple of str
    - range of str
    - slice of str
- `channel_indices` will be 0-based **(Needs to be clearly documented)**
- Accepted range of `channel_indices`: [0 : (num_channels_per_instrument * num_instruments) - 1]
- Return value will be `list` of `str` representing fully-qualified channel names like "PXI1Slot2/0,PXI1Slot5/31" on multi-instrument sessions and channel numbers like "0,1,2" on single-instrument sessions

`get_channel_name_from_string()` and `get_channel_name()` will be removed from nidigital Python bindings.

Implementation of `get_channel_names(channel_indices)` in nidigital will use `niDigital_GetChannelNameFromString()` C entry-point and converters for handling the different types of input and for converting comma-delimited string to list on return value.

NI-Scope C API also contains functions equivalent to `get_channel_name_from_string()` and `get_channel_name()`, but niscope Python bindings doesn't. Similar to nidigital, `get_channel_names(channel_indices)` will be added to niscope - #1402.

nidcpower, nifgen, niswitch Python bindings already have `get_channel_name()`, whose input is a single 1-based channel index. `get_channel_name()` will be deprecated and `get_channel_names(channel_indices)` (with the same behavior as explained above) will be added.

PR #1410 (2020-04-17): Improve usability of getting channel names from channel indices in nidigital
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

- Add get_channel_names()
- Remove get_channel_name() and get_channel_name_from_string()

### List issues fixed by this Pull Request below, if any.

* Fix #1386

### What testing has been done? [... cut]
closed 2020-04-17 by marcoskirsch