ni/nimi-python #1389: Provide string representation for data returned by fetch_capture_waveform() in nidigital

sbethur, 2020-04-08:
### Description of issue

`fetch_capture_waveform` returns a dictionary:
https://github.com/ni/nimi-python/blob/8540e86f00dc4b7aa9dbf5d905275db7255d9c7b/generated/nidigital/nidigital/session.py#L1252

The documentation claims the dict values are of type `array.array` but the method returns `memoryview`. More importantly, the default string representation of `memoryview` does not return the data, which makes it hard for users to print the returned data or save it to file.

`memoryview` is used in the implementation because underlying C entry-point returns a flat buffer containing data for all the sites. In Python, `memoryview` objects are created for slices within the buffer and returned.

sbethur, 2020-04-08:
I see two options:

### Option 1
Modify the API to return `list` or `array.array` instead of `memoryview`. We can then leverage the default string representation provided. The downside is the extra copy that would be incurred.

### Option 2
Modify the API to return a custom type:
```
class CaptureWaveformData():
    def __init__(self, data):
        self.data = data

    def __repr__(self):
        return str(self)

    def __str__(self):
        return str(self.data.tolist())
```

Extra copy would be avoided if the `data` member of `CaptureWaveformData` is used. Returned dictionary will support nicer string representation:
```
>>> cap_data = memoryview(array.array('L', [1, 2, 3]))
>>> waveforms = {0: CaptureWaveformData(cap_data)}
>>> print(waveforms[0])
[1, 2, 3]
>>> print(waveforms)
{0: [1, 2, 3]}
>>> print(type(waveforms[0].data))
<class 'memoryview'>
```

thejcannon, 2020-04-17:
You could document as returning a sequence, then use Option 2, but provide the full sequence protocol?

sbethur, 2020-04-24:
Discussed offline and decided none of the above two options are better than not doing anything and returning the capture waveform data as a dictionary of `memoryview` instances.

Option 1 is not good because of the extra copy.
Option 2 is not good because of the complexity associated with adding a wrapper that implements full sequence protocol, just for getting a better string representation of the returned object. See #1430 for details about the incurred complexity.

Returning a  dictionary of `memoryview` instances seems like the right compromise - No extra copy and no custom wrapper. Getting better string representation is trivial:
```
fetched_waveforms = session.fetch_capture_waveform()
print({k: v.tolist() for k, v in fetched_waveforms.items()})
```

>The documentation claims the dict values are of type `array.array` but the method returns `memoryview`.

This was fixed in #1427

closed 2020-04-24 by sbethur