ni/nimi-python #1392: Improve usability of loading specifications, levels, and timing files in nidigital

sbethur, 2020-04-09:
### Description of issue

Users typically load multiple levels files, multiple timing files, and at least one specifications file. Currently this requires `load_levels()`, `load_timing()`, and `load_specifications()` to be called in separate loops.

Usability can be improved by exposing a single method that allows multiple files of all three kinds to be loaded, similar to [Load Specifications Levels and Timing](https://www.ni.com/documentation/en/ni-digital/latest/ni-digital-node-ref/load-spec-levels-timing-multiple/) in LV.

marcoskirsch, 2020-04-09:
This ask doesn't make the API more Pythonic, it's just an improvement that would be equally applicable to LabVIEW and C and .NET APIs.

So I don't think it's a good idea to add it as a Python-specific deviation.

sbethur, 2020-04-09:
It won't be Python-specific. We have these as Poly VIs in LabVIEW, which are very similar to what I'm proposing in Python. In .NET, we have them in slightly different form - 3 different methods for specs, levels, and timing; each capable of loading one or more files. It had to be done like that in .NET because language doesn't support multiple optional parameters. If anything, C is the outlier; which is the case for many other parts of the API too.

PR #1394 (2020-04-14): Improve usability of loading specifications, levels, and timing files and unloading specifications
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

Improve usability of loading specifications, levels, and timing files and unloading specifications:
- Added `load_specifications_levels_and_timing` that allows loading of  [... cut]
closed 2020-04-14 by marcoskirsch