ni/nimi-python #1463: niscope module does not expose calibration info functionality

marcoskirsch, 2020-05-21:
The following functions exist in NI-SCOPE runtime and can be accessed through a regular (not calibration) session. Yet they are not exposed in `nimi-python`. Equivalent methods are exposed in `nimi-python` for other driver modules. 

* niScope_CalFetchDate
* niScope_CalFetchTemperature
* niScope_CalFetchCount
* niScope_CalFetchMiscInfo

When adding to nimi-python, we should make sure they are exposed in a way that is consistent with the other drivers in order to allow duck-typing usage.

NI-SCOPE uses an enum to select between self/external/manufacturing. We probably would be ok exposing self/external only as separate methods like all the other APIs do.

marcoskirsch, 2020-05-21:
Related issue: #1462

marcoskirsch, 2020-05-21:
Internal Bug 1038236: A customer ran into this issue and needs the cal date functionality exposed.

shilohbeeler-NI, 2020-05-22:
> niScope_CalFetchCount
> niScope_CalFetchMiscInfo
Could you point me to an equivalent implementation of either of these two in another module? I'm unable to find any. I've got the other two handled but I don't want to try writing the metadata for these two from scratch unless necessary.

marcoskirsch, 2020-05-23:
No other module has anything like `niScope_CalFetchCount` because none of the other driver runtimes have equivalent functionality as far as I can tell. I am not sure if we should add this to `nimi-python` – we need to research a bit more. For instance: is this something supported on current modules or only on EOL ones?

`niScope_CalFetchMiscInfo` is equivalent to:

* `niDCPower_GetCalUserDefinedInfo`
* `niDMM_GetCalUserDefinedInfo`
* `niFgen_GetCalUserDefinedInfo`

Why is NI-SCOPE inconsistent? Great question.
We should expose all four consistently in `nimi-python` in my opinion because Duck Typing.

PR #1469 (2020-06-11): NI-SCOPE: Expose calibration datetime and temperature functions in the API.
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

Exposes niScope_CalFetchDate, niScope_CalFetchTemperature, and ~~niScope_CalFetchMiscInfo~~ in a manner consistent with other modules.

### List issues fixed by this Pull Request below, if any.
 [... cut]
closed 2020-06-11 by marcoskirsch