ni/nimi-python #1473: Unintentional dependency on pytest?

tmhofler, 2020-06-01:
### Description of issue

nidcpower has a dependency on pytest (I assume unintentional since it is not listed as a dependency).

### System report

`python -c "import niscope; niscope.print_diagnostic_information()"` output (replace `niscope` with appropriate package name)

python -c "import nidcpower; nidcpower.print_diagnostic_information()
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\nitest\.virtualenvs\nidcpower_test-yBwmuPo5\lib\site-packages\nidcpower\__init__.py", line 10, in <module>
    from nidcpower.session import Session  # noqa: F401
  File "C:\Users\nitest\.virtualenvs\nidcpower_test-yBwmuPo5\lib\site-packages\nidcpower\session.py", line 8, in <module>
    import nidcpower._attributes as _attributes
  File "C:\Users\nitest\.virtualenvs\nidcpower_test-yBwmuPo5\lib\site-packages\nidcpower\_attributes.py", line 3, in <module>
    import nidcpower._converters as _converters
  File "C:\Users\nitest\.virtualenvs\nidcpower_test-yBwmuPo5\lib\site-packages\nidcpower\_converters.py", line 10, in <module>
    import pytest
ModuleNotFoundError: No module named 'pytest'

### Steps to reproduce issue

`pip install nidcpower==1.3.0`
`python`
`import nidcpower`
> Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\nitest\.virtualenvs\nidcpower_test-yBwmuPo5\lib\site-packages\nidcpower\__init__.py", line 10, in <module>
    from nidcpower.session import Session  # noqa: F401
  File "C:\Users\nitest\.virtualenvs\nidcpower_test-yBwmuPo5\lib\site-packages\nidcpower\session.py", line 8, in <module>
    import nidcpower._attributes as _attributes
  File "C:\Users\nitest\.virtualenvs\nidcpower_test-yBwmuPo5\lib\site-packages\nidcpower\_attributes.py", line 3, in <module>
    import nidcpower._converters as _converters
  File "C:\Users\nitest\.virtualenvs\nidcpower_test-yBwmuPo5\lib\site-packages\nidcpower\_converters.py", line 10, in <module>
    import pytest
ModuleNotFoundError: No module named 'pytest'

sbethur, 2020-06-01:
Yup, it's an unintentional dependency :( We have some unit tests for `converters`. In a recent change some unit tests started using `pytest.approx`. I see couple of things we need to address:

1. Remove the pytest dep and do a patch release - Should be straight-forward to do.
2. Ensure we don't add unintentional deps used by unit tests in the future - The unit tests are run only on `nifake` module. I propose we put the unit tests in converters mako under `% if config['module_name'] == 'nifake':`. Then `flake8` can catch unused deps in the generated converters file.
3. Create installer test to catch unintentional deps. Not clear how we can do it. But there is an existing issue to address this - https://github.com/ni/nimi-python/issues/1446. We should prioritize this.

@marcoskirsch, thoughts?

marcoskirsch, 2020-06-01:
> I propose we put the unit tests in converters mako under % if config['module_name'] == 'nifake':. Then flake8 can catch unused deps in the generated converters file.

Or move them to their own separate file so that we don't add further conditional logic to the mako template.

thejcannon, 2020-06-02:
Having the tests be in the same file as the source code is not something I'd expect when installing a Python module.

- It pollutes the API of the module (attribute discovery will show a bunch of unrelated functions)
- It can pollute the module's dependencies
- It bloats the installed modules (more LOC to download and sift through)
- It makes features such as `pytest`'s `conftest.py` harder to use.'

If you want people to be able to run tests after installation you should allow installation using an extra (like `nidmm[test]`)

marcoskirsch, 2020-06-02:
> If you want people to be able to run tests after installation you should allow installation using an extra (like nidmm[test])

We don't.

marcoskirsch, 2020-06-02:
> is not something I'd expect when installing a Python module.

Meeting Python user expectations is important to `nimi-python` so this is good feedback.

PR #1477 (2020-06-05): Split the converter functions and its unit tests into two files
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

~- [ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~

~- [ ] I've added tests applicable for this pull request~

### What does this Pull Request accomplish?

Converter functions and unit tests for them were in the same file. This is Unpythonic as explained in #1475. It's also a maintenance headache as seen in #1473.

I've split the functions and unit tests i [... cut]
closed 2020-06-05 by marcoskirsch