ni/nimi-python #1475: Test code should be separate from module code

thejcannon, 2020-06-02:
(Copied from #1473)

Having the tests be in the same file as the source code is not something I'd expect when installing a Python module.

- It pollutes the API of the module (attribute discovery will show a bunch of unrelated functions)
- It can pollute the module's dependencies
- It bloats the installed modules (more LOC to download and sift through)
- It makes features such as `pytest`'s `conftest.py` harder to use.

If you want people to be able to run tests after installation you should allow installation using an extra (like `nidmm[test]`)

sbethur, 2020-06-03:
#1477 should fix this issue for `converters`. With that, none of the "shipping" modules should have this problem.
But there are other modules in [helper folder](https://github.com/ni/nimi-python/tree/master/build/helper), which are used during codegen, that we need to fix.

closed 2023-05-08 by marcoskirsch

thejcannon, 2023-05-09:
:rocket:

PR #1961 (2023-05-08): Split build unit tests out from code
- [X] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- ~[ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~
- [X] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

Stop keeping the unit tests for build/helper together with the code. It messes up coverage analysis, bloats the helper files and often makes the tests bigger and uglier because we aren't using features like pytest.mark.parameterize.
 [... cut]