ni/nimi-python #1486: Need to access external calibration date and time

jperrywinters, 2020-07-08:
### Description of issue

    For some other instruments there are methods to access the date/time of the last external calibration (such as "get_ext_cal_last_date_and_time()" for niscope).  I cannot find any means to access this information for the PXIe-6570.  It exists in memory somewhere because the external calibration date can be seen when using NI-Max.  Is there a way to add this functionality to the nimi-python drivers for nidigital?

[image]

marcoskirsch, 2020-07-13:
The NI-Digital runtime doesn't provide a way to get te external calibration date and time. See attached C header file (extension changed to TXT because of GitHub restrictions).

I suppose customers get around this limitation by calling [NI System Configuration API](https://forums.ni.com/t5/Community-Documents/Welcome-to-the-NI-System-Configuration-API/ta-p/3526512?profile.language=en). But that's not helpful to you since there is no Python support for that API :(

[niDigital.h.txt](https://github.com/ni/nimi-python/files/4912510/niDigital.h.txt)

I will pass this feedback along.

jperrywinters, 2020-07-13:
Hi Marcos,

That is unfortunate to hear, but thank you very much for the reply.  Would it be correct to say, then, that this is a firmware/software limitation for NI-Digital instruments?  Would you be able to point me towards somewhere I could make this feature request for that type of device?

Thanks again,

-Josh

marcoskirsch, 2020-07-13:
It's a limitation of the NI-Digital Pattern Driver API. As you correctly observed, the information is visible in MAX which means that there is a way to retrieve it from hardware.

The way the MAX UI works in the NI-Digital Pattern Driver case, it calls into NI System Configuration API. Without getting into too much detail, the NI-Digital driver has a plug-in for NI System Configuration API that implements all it needs for the specific hardware. So all the low-level logic is there in the driver, it just isn't exposed in the API, and thus nimi-python cannot expose it.

The lack of System Configuration API for Python is a known gap.

marcoskirsch, 2020-07-21:
Closing, since this is not a `nimi-python` issue.

closed 2020-07-21 by marcoskirsch