ni/nimi-python #1491: import_attribute_configuration_buffer() fails intermittently when `list` or `array.array` is passed in

sbethur, 2020-07-22:
### Description of issue

[import_attribute_configuration_buffer](https://github.com/ni/nimi-python/search?q=%22import_attribute_configuration_buffer%28self%2C+configuration%29%3A%22&unscoped_q=%22import_attribute_configuration_buffer%28self%2C+configuration%29%3A%22) fails intermittently when the passed-in `configuration`'s type is `list` of `int` or `array.array('b')`.

### System report

Command:
```
\python.exe -c "import nidcpower; nidcpower.print_diagnostic_information()"
```

Output:
```
OS:
    Name:      Windows
    Version:   10.0.17763
    Bits:      64
Driver:
    Name:      NI-DCPower
    Version:   19.1.0.49152
Module:
    Name:      nidcpower
    Version:   1.3.1
Python:
    Version:   3.6.0 (v3.6.0:41df79263a11, Dec 23 2016, 08:06:12) [MSC v.1900 64 bit (AMD64)]
    Bits:      64
    Is_Venv:   False
    Installed Packages:
        setuptools==28.8.0
        pip==9.0.1
        nidcpower==1.3.1
        hightime==0.2.0
```

### Steps to reproduce issue

Run the following script few times (Hard to say how many times to run before it fails, since is related to accessing invalid memory locations) and see the error:
```
import nidcpower

with nidcpower.Session('SMU1', '0', False, 'Simulate=1, DriverSetup=Model:4135; BoardType:PXIe') as session:
    # data is  generated by running the following:
    #    print(session.export_attribute_configuration_buffer())
    data = b'{\n   "formatVersion": 2,\n   "oldestCompatibleFormatVersion": 2,\n   "timeStamp": "2020-07-14T21:25:22.728466",\n   "session": {\n      "resourceName": "__ivi_sim1",\n      "channels": "0",\n      "optionString": "Simulate=1, DriverSetup=Model:4135; BoardType:PXIe"\n   },\n   "generatedBy": {\n      "name": "nidcpowerPAL_aux.dll",\n      "version": "20.1.0a4"\n   },\n   "devices": [\n      {\n         "alias": "__ivi_sim1",\n         "modelNumber": 2015282372\n      }\n   ],\n   "attributes": [\n      {\n         "id": 1150007,\n         "name": "NIDCPOWER_ATTR_OVERRANGING_ENABLED",\n         "channel": "",\n         "value": false\n      },\n      {\n         "id": 1150008,\n         "name": "NIDCPOWER_ATTR_OUTPUT_FUNCTION",\n         "channel": "0",\n         "value": 1006\n      },\n      {\n         "id": 1150013,\n         "name": "NIDCPOWER_ATTR_SENSE",\n         "channel": "0",\n         "value": 1008\n      },\n      {\n         "id": 1150015,\n         "name": "NIDCPOWER_ATTR_VOLTAGE_LEVEL_AUTORANGE",\n         "channel": "0",\n         "value": 1\n      },\n      {\n         "id": 1150016,\n         "name": "NIDCPOWER_ATTR_CURRENT_LIMIT_AUTORANGE",\n         "channel": "0",\n         "value": 1\n      },\n      {\n         "id": 1150020,\n         "name": "NIDCPOWER_ATTR_POWER_LINE_FREQUENCY",\n         "channel": "0",\n         "value": 60.0\n      },\n      {\n         "id": 1150021,\n         "name": "NIDCPOWER_ATTR_START_TRIGGER_TYPE",\n         "channel": "",\n         "value": 1012\n      },\n      {\n         "id": 1150023,\n         "name": "NIDCPOWER_ATTR_DIGITAL_EDGE_START_TRIGGER_INPUT_TERMINAL",\n         "channel": "",\n         "value": ""\n      },\n      {\n         "id": 1150030,\n         "name": "NIDCPOWER_ATTR_SOURCE_TRIGGER_TYPE",\n         "channel": "",\n         "value": 1012\n      },\n      {\n         "id": 1150032,\n         "name": "NIDCPOWER_ATTR_DIGITAL_EDGE_SOURCE_TRIGGER_INPUT_TERMINAL",\n         "channel": "",\n         "value": ""\n      },\n      {\n         "id": 1150033,\n         "name": "NIDCPOWER_ATTR_EXPORTED_SOURCE_TRIGGER_OUTPUT_TERMINAL",\n         "channel": "",\n         "value": ""\n      },\n      {\n         "id": 1150034,\n         "name": "NIDCPOWER_ATTR_MEASURE_TRIGGER_TYPE",\n         "channel": "",\n         "value": 1015\n      },\n      {\n         "id": 1150037,\n         "name": "NIDCPOWER_ATTR_EXPORTED_MEASURE_TRIGGER_OUTPUT_TERMINAL",\n         "channel": "",\n         "value": ""\n      },\n      {\n         "id": 1150043,\n         "name": "NIDCPOWER_ATTR_SOURCE_COMPLETE_EVENT_OUTPUT_TERMINAL",\n         "channel": "",\n         "value": ""\n      },\n      {\n         "id": 1150047,\n         "name": "NIDCPOWER_ATTR_MEASURE_COMPLETE_EVENT_OUTPUT_TERMINAL",\n         "channel": "",\n         "value": ""\n      },\n      {\n         "id": 1150051,\n         "name": "NIDCPOWER_ATTR_SOURCE_DELAY",\n         "channel": "0",\n         "value": 0.0\n      },\n      {\n         "id": 1150057,\n         "name": "NIDCPOWER_ATTR_MEASURE_WHEN",\n         "channel": "",\n         "value": 1027\n      },\n      {\n         "id": 1150058,\n         "name": "NIDCPOWER_ATTR_APERTURE_TIME",\n         "channel": "0",\n         "value": 1.0\n      },\n      {\n         "id": 1150059,\n         "name": "NIDCPOWER_ATTR_APERTURE_TIME_UNITS",\n         "channel": "0",\n         "value": 1029\n      },\n      {\n         "id": 1150062,\n         "name": "NIDCPOWER_ATTR_TRANSIENT_RESPONSE",\n         "channel": "0",\n         "value": 1038\n      },\n      {\n         "id": 1150063,\n         "name": "NIDCPOWER_ATTR_MEASURE_RECORD_LENGTH",\n         "channel": "0",\n         "value": 1\n      },\n      {\n         "id": 1150095,\n         "name": "NIDCPOWER_ATTR_PULSE_TRIGGER_TYPE",\n         "channel": "",\n         "value": 1012\n      },\n      {\n         "id": 1150184,\n         "name": "NIDCPOWER_ATTR_COMPLIANCE_LIMIT_SYMMETRY",\n         "channel": "0",\n         "value": 0\n      },\n      {\n         "id": 1250001,\n         "name": "NIDCPOWER_ATTR_VOLTAGE_LEVEL",\n         "channel": "0",\n         "value": 0.0\n      },\n      {\n         "id": 1250005,\n         "name": "NIDCPOWER_ATTR_CURRENT_LIMIT",\n         "channel": "0",\n         "value": 0.1\n      },\n      {\n         "id": 1250006,\n         "name": "NIDCPOWER_ATTR_OUTPUT_ENABLED",\n         "channel": "0",\n         "value": false\n      },\n      {\n         "id": 1150054,\n         "name": "NIDCPOWER_ATTR_SOURCE_MODE",\n         "channel": "",\n         "value": 1020\n      }\n   ],\n   "advancedSequences": []\n}\x00'
    session.import_attribute_configuration_buffer(data)
    data_as_list = list(data)
    session.import_attribute_configuration_buffer(data_as_list) # This throws intermittently with "data is invalid" error
```

Exception thrown:
```
Traceback (most recent call last):
  File "C:\Users\admin\Desktop\dcpowerIssue\dcpowerIssue.py", line 10, in <module>
    session.import_attribute_configuration_buffer(data_as_list) # This throws intermittently with "data is invalid" error
  File "C:\Users\admin\AppData\Local\Programs\Python\Python36\lib\site-packages\nidcpower\session.py", line 73, in aux
    return f(*xs, **kws)
  File "C:\Users\admin\AppData\Local\Programs\Python\Python36\lib\site-packages\nidcpower\session.py", line 4783, in import_attribute_configuration_buffer
    errors.handle_error(self, error_code, ignore_warnings=False, is_error_handling=False)
  File "C:\Users\admin\AppData\Local\Programs\Python\Python36\lib\site-packages\nidcpower\errors.py", line 95, in handle_error
    raise DriverError(code, description)
nidcpower.errors.DriverError: -1074100298: Unable to import the attribute configuration. The input data is invalid.
```

sbethur, 2020-07-22:
### Root-cause analysis

The passed-in `configuration` object is converted to ctypes object like this: https://github.com/ni/nimi-python/blob/b624a4be4e1988539db6769e84c35116e2a10557/generated/nidcpower/nidcpower/session.py#L4781

which boils down to:
```
configuration_ctype = ctypes.cast(obj=bytes(configuration), type=ctypes.POINTER(ctypes.c_int8))
```

According to [documentation](https://docs.python.org/3/library/ctypes.html#ctypes.cast) of `ctypes.cast`:
> obj must be an object that can be interpreted as a pointer.

In the generated code, reference to temporary object is passed into to `obj` parameter, which is destructed after the `cast` statement executes.

The below code snippets show the problem
Correct usage:
```
import ctypes

class B(bytes):
    def __repr__ (self):
        return 'Instance of B'

    def __del__(self):
        print('Deleting %r' % self)
       
b = B(b'foo')
cobj = ctypes.cast(b, ctypes.POINTER(ctypes.c_int8))
print('Casted ctypes object will be used here')
print(ctypes.string_at(cobj))
print('End of the script')
```

Output:
```
Casted ctypes object will be used here
b'foo'
End of the script
Deleting Instance of B
```

Wrong usage:
```
import ctypes

class B(bytes):
    def __repr__ (self):
        return 'Instance of B'

    def __del__(self):
        print('Deleting %r' % self)
       
cobj = ctypes.cast(B(b'foo'), ctypes.POINTER(ctypes.c_int8))
print('Casted ctypes object will be used here')
print(ctypes.string_at(cobj))
print('End of the script')
```

Output:
```
Deleting Instance of B
Casted ctypes object will be used here
b'l be used here'
End of the script
```

As you can see in the output of the wrong usage, the object is destructed before it's used and the ctypes pointer to points to invalid memory location.

BTW, if `bytes` object is passed in, then..
```
configuration_ctype = ctypes.cast(obj=bytes(configuration), type=ctypes.POINTER(ctypes.c_int8))
```
..the call to bytes() will just increment the refcount on the passed in object rather than creating a new temporary object. So, nothing is broken.

The fix is pretty simple. Change to:
```
configuration_as_bytes = _converters.convert_to_bytes(configuration)
configuration_ctype = get_ctypes_pointer_for_buffer(value=configuration_as_bytes, library_type=_visatype.ViInt8)  # case B520
```

marcoskirsch, 2020-07-22:
So just extend the lifetime of `configuration_as_bytes`...
Hopefully the changes to the generator are straightforward!

PR #1493 (2020-07-28): Fix intermittent failure in import_attribute_configuration_buffer()
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [x] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.

~- [ ] I've added tests applicable for this pull request~

### What does this Pull Request accomplish?

Fix intermittent failure in import_attribute_configuration_buffer() by updating code generator to pass non-temporary object to `get_ctypes_pointer_for_buffer()`.

### List issues fixed by this Pull Re [... cut]
closed 2020-07-28 by marcoskirsch