ni/nimi-python #1497: meas_other_channel property should accept all types supported by channel repeated capability

sbethur, 2020-08-05:
### Description of issue

Property in question:
https://github.com/ni/nimi-python/blob/df494361119cab7bc27191cef77925661cfe03c7/generated/niscope/niscope/session.py#L902

Currently, the property only supports `str` type when setting. But it should support other types (like int) that are supported when channel is passed in as repeated capability. Code like this should be made to work:
```
my_session.meas_other_channel = 2
```

sbethur, 2020-08-10:
I am thinking of creating a new attribute [class](https://github.com/ni/nimi-python/blob/master/generated/niscope/niscope/_attributes.py):
```
class AttributeViStringRepeatedCapability(Attribute):

    def __get__(self, session, session_type):
        return session._get_attribute_vi_string(self._attribute_id)

    def __set__(self, session, value):
        session._set_attribute_vi_string(self._attribute_id, _converters.convert_repeated_capabilities_without_prefix(value))
```

` _converters.convert_repeated_capabilities_without_prefix()` is called on set value before calling the `set_attr()` method. This attribute class would then be useful in cases where property represents a repeated capability based types like channels, instruments etc.

For `meas_other_channel`, it would allow the users to set it to an `int`.

@marcoskirsch. does this sound good?

marcoskirsch, 2020-08-10:
I think this is consistent with how we solved other attribute "types".
There is a question on how to handle this in metadata, but it is interesting API metadata to have in general. For example, we have attributes whose type is string, but in reality it is "string representing a routable trigger terminal". That can inform APIs beyond Python.

sbethur, 2020-08-11:
>For example, we have attributes whose type is string, but in reality it is "string representing a routable trigger terminal". That can inform APIs beyond Python.

That's a good point. When we have a better idea of how we want to want to use that metadata, we should consider making them generic. For now, I'm planning to add `attribute_class` key to Python-specific metadata, like existing instances:
```
'attribute_class': 'AttributeViReal64TimeDeltaSeconds',
```

sbethur, 2020-08-20:
#1506 setup the infrastructure to fix this issue but there's more to be done.

PR #1506 (2020-08-20): Add support for properties of type `str` that represents a repeated capability
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

~- [ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~

- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

- Add support for properties of type `str` that represents a repeated capability
- It will be used to fix #1497 

### List issues fixed by this Pull Request below, if any.

Contribute towar [... cut]
commit f4d44376 (2020-08-20): Add support for properties of type `str` that represents a repeated capability (#1506)

* Add support for properties of type `str` that represents a repeated capability

It will be used to fix #1497.

* Update documentation for the new property
PR #1511 (2020-08-21): Make meas_other_channel in niscope work with channel repeated capability types
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

~- [ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~

~- [ ] I've added tests applicable for this pull request~

### What does this Pull Request accomplish?

Make meas_other_channel in niscope work with channel repeated capability types. Users can now pass in `str` or `int` or any other type supported by channel repeated capability.

Metadata  [... cut]
closed 2020-08-21 by marcoskirsch