ni/nimi-python #1501: Follow up on purpose of add_waveform_processing in NI-SCOPE.

shilohbeeler-NI, 2020-08-07:
The purpose of the add_waveform_processing method in NI-SCOPE is a little unclear -- it's possible it might just be an alternate way to manually use fetch_array_measurements in conjunction with other methods. If so, we should look into deciding which method should be made the single way of fetching processed array measurements.

sbethur, 2020-08-10:
If we decide to keep `fetch_array_measurement`, we need to decide whether it should include `measWfmSize` parameter, like in C API.

sbethur, 2020-08-11:
`add_waveform_processing` allows registering multiple measurement steps to be done on the waveform, before fetching the array measurement using `fetch_array_measurement`.
`fetch_array_measurement` allows specifying a single array measurement function.

It seems like we can provide a fancy method that makes the waveform processing atomic, where users can specify one or more measurement steps and get back the array measurement in a single call. Internally, it will:
- Call `add_waveform_processing`, as many times as required
- Call `fetch_array_measurement` to get the measurement
- Finally call `clear_waveform_processing`

Couple of issues I see with the above fancy method:
- `add_waveform_processing` has side-effects outside of `fetch_array_measurement`, i.e. on `fetch_measurement` and `fetch_measurement_stats`. Theoretically users can add some waveform processing make one ore more calls to `fetch_measurement` before clearing the waveform processing. With the fancy method, they would not be able to do it.
- If you need to apply the same waveform processing for multiple `fetch_array_measurement`s, then they get applied and cleared multiple times with the fancy method. This could be bad from performance standpoint.

I'm leaning towards keeping the methods separate, i.e. have these methods in the API:
- `add_waveform_processing`
- `fetch_array_measurement`
- `clear_waveform_processing`

(I want to get consensus on the above before thinking about `measWfmSize` parameter.)

homechicken80, 2020-08-11:
I agree with keeping the methods separate.  The processing steps are advanced features that only advanced customers use.  Also keep in mind that some attributes are also stored when adding a processing step (in that way the attributes are like list mode).  Therefore combining the functions doesn't enable many advanced use cases.

sbethur, 2020-08-11:
@homechicken80, thanks for the response. One usability improvement we could do is to make `measWfmSize` parameter optional  with Default Value: `None` (-1 when calling driver runtime) (returns all available samples). This will match the LabVIEW API and is consistent with what we've done for `fetch` and `read` in nimi-python. Does that sound good?

homechicken80, 2020-08-11:
If I remember correctly that would depend on your implementation.  '-1' means all the samples which may be less than buffer allocation (in c).  In LabVIEW we reshape the array based on the actual samples returned.  If you are handling all the memory allocation then make sure to handle the actual buffer returned (to prevent uninitialized data being returned to the customer).

marcoskirsch, 2020-08-12:
@sbethur we need to remember to test all the different memory allocation options in the automated tests.

sbethur, 2020-08-13:
Continuing the discussion on design of `fetch_array_measurement`.. I noticed that `fetch`, `fetch_into`, and `read` allow users to pass in some properties related to Acquisition as params:
https://github.com/ni/nimi-python/blob/9c8e3967311f470e637c84fe042bcf36b200433f/generated/niscope/niscope/session.py#L1909

Assuming those properties apply to `fetch_array_measurement` and `fetch_measurement_stats` as well, I think we should do the same for these methods too.

@marcoskirsch @homechicken80 @ni-balexand , what do you guys think?

homechicken80, 2020-08-13:
That's kind of unfortunate we did that.  While it is true we added those attributes so that we didn't have to create new function signatures; hiding the implementation by including the parameters in the function call probably isn't a good idea.  Are we sure we're including that parameter on every function that might be using it under the hood?  ... points done/fetch backlog?

Anyway, unless the 'fetch' call is returning the offset/relativeTo parameters back to their precall value I think you should (have to?) include them on the fetch array and stats calls as parameters.

Which is channel in the function call?  Is this a per channel call?  Remember that these parameters impact the other channel.  In specific fetching relative to read pointer will automatically advance the read pointer (and there is no way to rewind it).  

You may or may not care, but reading channels in separate calls on some devices (like the 5105) could take N times longer.

sbethur, 2020-08-13:
>Are we sure we're including that parameter on every function that might be using it under the hood? ... points done/fetch backlog?

I'm not sure. I see `points_done` and `backlog` are plain-old simple properties. I don't see any function for them.

I can elaborate on what we do with fetch methods:
In addition to the params in the C function, we have:
- `relative_to` (default value `enums.FetchRelativeTo.PRETRIGGER`)
- `offset` (default value `0`)
- `record_number` (default value `0`)
- `num_records` (default value `-1`)

The fetch methods accept channels as repeated capability, so for channelList = "0,1", you'd call like:
```
scope_session.channels[0,1].fetch(num_samples=1000, num_records=3)
```

In the Python layer, before calling `niScope_Fetch()`, we do:
```
        # Set the fetch attributes
        with _NoChannel(session=self):
            self._fetch_relative_to = relative_to
            self._fetch_offset = offset
            self._fetch_record_number = record_number
            self._fetch_num_records = -1 if num_records is None else num_records
            if num_samples is None:
                num_samples = self.horz_record_length
```
(The `_NoChannel` is to call `niScope_SetAttributeXYZ()` with an empty `channelList` since they are not channel-based)

> Anyway, unless the 'fetch' call is returning the offset/relativeTo parameters back to their precall value I think you should (have to?) include them on the fetch array and stats calls as parameters.

`offset` and `relative_to` parameters passed to `fetch` are input parameters, so their values don't change when `fetch` is called. The properties corresponding to them are not updated by `fetch` after it calls `niScope_Fetch()`; so they are not reset to precall values.

sbethur, 2020-09-08:
@includekunal, I've created a PR - #1518 that addresses the issues discussed here. Could you please review it.

includekunal, 2020-09-09:
Is your intention that setting this parameters will persistently impact other attributes/functions (since the under the hood you set the attributes)?  That behavior seems to already exist for the fetch functions, but I want to make sure I understand the behavior.

sbethur, 2020-09-09:
> Is your intention that setting this parameters will persistently impact other attributes/functions (since the under the hood you set the attributes)? That behavior seems to already exist for the fetch functions, but I want to make sure I understand the behavior.

Yes, that is correct.

PR #1518 (2020-09-09): Add additional fetch related parameters to `fetch_array_measurement` and `fetch_measurement_stats` methods
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

~- [ ] I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~

- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

Add additional fetch related parameters to `fetch_array_measurement` and `fetch_measurement_stats` methods. This makes these methods be consistent with `fetch`:
 [... cut]
closed 2020-09-09 by marcoskirsch