ni/nimi-python #1517: collaboration with instrument studio not working...

leokleingmx, 2020-09-08:
I tried using Instrument Studio during debugging a python program and it cannot hand over the session (?). Basically the same worked quite fine with .NET...

### Description of issue
PXIe-4112 (SMU-POWER) access is lost between python and instrument Studio. I had to restart the machine to get it back.
- Instrument Studio build: 19.0.0.49152
- nimi-python package installed today.
- Visual Studio 2019

### Steps to reproduce issue
Code:
``` python
import nidcpower
with nidcpower.Session(resource_name='PXI4112_SUPPLY2', channels='0') as session:
# breakpoint here!
    session.measure_record_length = 20
    session.commit()
```
1. Start VisualStudio debugging
2. Stop at breakpoint
3. Start Instrument Studio -> Debug. It finds the opened instrument.
4. After that Python cannot execute "nidcpower.Session(resource_name='PXI4112_SUPPLY2', channels='0')" and breaks with exeption: "-1074097933: Amount of time allocated to perform this operation was exceeded."
5. Instrument Studio will also find out: "Unable to connect to the external session. This could be a result of setting a breakpoint in your host application. Select Configure Debug Session from the menu in the instrument sheader to enable debugging with breaktpoints." This setting is already set before, but the menu item is greyed out.
6. Restart the system to get back the function. I tried in MAX to reset the instrument, but that doesn't help

Maybe I'm missing it but I was expecting Instrument Studio to work together with nimi-python?

leokleingmx, 2020-09-09:
### Further findings
- It seems to happen only when using the option : "Using Breakpoints in C/C++/.NET Applications" in Instrument studio
- It will break the whole class (eg. all nidmm instruments, or all nidcpower (here: SMUs and Power instruments).

- It seems to be a instrument studio problem(?)

sbethur, 2020-09-09:
I think this is the same issue as the one reported in #906. @leokleingmx, the workaround for now is to disable the "Using Breakpoints in C/C++/.NET Applications" option. Please check if that fixes the problem.

leokleingmx, 2020-09-10:
Thanks sbethur. You're right. Clearing the checkbox fixes the problem.
I couldn't link my problem to the other issue although I've seen it... Maybe this issue title will then help others...

Closed and I hope that the CAR 717410 will soon be fixed.

closed 2020-09-10 by leokleingmx