ni/nimi-python #1528: Publish nidigital 1.0.0 on PyPI

sbethur, 2020-10-23:
### Description of issue

Publish nidigital 1.0.0 on [PyPI](https://pypi.org/project/nidigital/)

ni-jxie, 2021-02-18:
@marcoskirsch @sbethur @phindman I believe this is the last issue we need to do for nidigital 1.0 Python release, hooray! When can we make this happen and celebrate?

marcoskirsch, 2021-02-18:
Very exciting. Good work.

#1540 and #1541 ... Do we want them in?

sbethur, 2021-02-19:
I have one PR left to submit for `nidigital`, which I'll do today.

If #1540 and #1541 can be submitted in 1-2 days, I can wait before doing the release. @gfisher-ni, do you have the bandwidth to complete these?

gfisher-NI, 2021-02-19:
@sbethur I intend on working on those today, so, hopefully, yes.

ni-jxie, 2021-02-25:
#1540 and #1541 are in! @sbethur Are we ready to do the release?

marcoskirsch, 2021-02-25:
Yep!
Expect it soon.

sbethur, 2021-03-02:
nidigital 1.0.0 has been released!!

https://github.com/ni/nimi-python/releases/tag/1.3.3

closed 2021-03-02 by sbethur