ni/nimi-python #1531: Many nidcpower attributes are documented as channel-based even though they are not.

alexdubois-ni, 2020-11-17:
### Description of issue

If you visit: https://nimi-python.readthedocs.io/en/master/nidcpower/class.html#properties, you can see that many properties are showing up as channel-based, even though they are channel-based. For instance, source mode is a non channel-based attribute but it is being marked as channel-based.

### System report

N/A

### Steps to reproduce issue

Step 1: Install NI-DCPower driver from ni.com
Step 2: Install nimi-python nidcpower
Step 3: Attempt to configure source mode with a repeated capability. It is not actually channel-based so it will fail.

sbethur, 2021-06-10:
@luisgomes252, did you get addressed in your metadata updates?

luisgomes252, 2021-06-10:
> @luisgomes252, did you get addressed in your metadata updates?

I don't think so. For nidcpower, I still allowed the internal metadata to determine the rep caps of each property. All I did was refactor the metadata tags themselves (bool to array, doc tables, etc), but I still allowed the prior hapigen content to determine what should be what.

As an example, I just now checked the source mode property that is referenced in this issue, and it's still marked as channel based. If this is wrong, then someone will still need to review each property and fix (in the hapigen files!) whatever is incorrect.

alexdubois-ni, 2021-06-10:
This should be fixed now. The problem was that these attributes were changed to channel-based when they were actually session-based due to not having independent channels support. Now that we’ve added it, these are now truly channel-based and thus the documentation is realigned with the actual driver. This should be fixed now.

closed 2021-06-18 by sbethur