ni/nimi-python #1532: Document Importing/Exporting Signals

gfisher-NI, 2020-12-01:
### Description of issue

As per [this pull request](https://github.com/ni/nimi-python/pull/848), the nimi-python APIs for several MI drivers no longer have the export_signal() function. This has led to some customer confusion, as this is a place where the Python APIs differ from the C APIs (where the corresponding ExportSignal() functions exist). Usually, if documentation doesn't exist for something in Python, users (and supporters) can reference the C or LabVIEW API and generally figure out what they need to do, but that's not the case here.

In cases like this, additional documentation noting this change and possibly showing an example of how to code the equivalent behavior should exist.

gfisher-NI, 2020-12-16:
For example, in either the documentation or the examples there is no indication of what values can be used for "digital_edge_start_trigger_source".

marcoskirsch, 2021-01-04:
The values that can be used change based on what instruments are in the system at any given time, so we cannot document it statically.

For example, you may want to set `digital_edge_start_trigger_source` to be `"/mySignalGenerator/MarkerEvent4"`. But this only works if there is a Signal Generator named "mySignalGenerator" on the system.

There is a larger that Python problem surrounding the documentation of our routing capabilities throughout driver software that will require a more concerted effort to resolve.