ni/nimi-python #1533: nidmm.Session.ConfigureOpenCableCompValues() is missing?

mturpin-pmc, 2020-12-07:
The docs for nidmm.Session.perform_open_cable_comp() says to use the returned values with nidmm.Session.ConfigureOpenCableCompValues(), but that is not available in nidmm.Session....   

> For the NI 4082 and NI 4072 only, performs the open cable compensation measurements for the 
> current capacitance/inductance range, and returns open cable compensation Conductance and 
> Susceptance values. You can use the return values of this method  as inputs to 
> nidmm.Session.ConfigureOpenCableCompValues()."

Thank you!

sbethur, 2020-12-08:
The documentation is misleading. Cable compensation can be configured using these properties:
- https://github.com/ni/nimi-python/blob/be1d5aa2eea7d22e223d64ae7012bf8e3ef79eea/generated/nidmm/nidmm/session.py#L151
- https://github.com/ni/nimi-python/blob/be1d5aa2eea7d22e223d64ae7012bf8e3ef79eea/generated/nidmm/nidmm/session.py#L267
- https://github.com/ni/nimi-python/blob/be1d5aa2eea7d22e223d64ae7012bf8e3ef79eea/generated/nidmm/nidmm/session.py#L273
- https://github.com/ni/nimi-python/blob/be1d5aa2eea7d22e223d64ae7012bf8e3ef79eea/generated/nidmm/nidmm/session.py#L351
- https://github.com/ni/nimi-python/blob/be1d5aa2eea7d22e223d64ae7012bf8e3ef79eea/generated/nidmm/nidmm/session.py#L357

sbethur, 2020-12-08:
@mturpin-pmc, thanks for bringing this to our attention. I've updated the title and tagged it as a documentation issue.