ni/nimi-python #1555: Numerous enumeration values lack documentation

phindman, 2021-02-04:
There are many enumeration values, such as PIN_STATE_NOT_ACQUIRED in nidigital, that do not have documentation.

_Originally posted by @sbethur in https://github.com/ni/nimi-python/pull/1552#discussion_r569897349_

phindman, 2021-02-10:
@sbethur I think the NI-Digital specific portion of this is complete.  There are still numerous enum values in other drivers (e.g. nidcpower.Event, nifgen.HardwareState, niscope.ClearableMeasurement) that should probably have documentation added, but don't need to be associated with the nidigital 1.0 milestone.  I'm not sure if you want to keep this open but remove it from the nidigital 1.0 milestone, or close it (and possibly open issues for the other driver APIs).

sbethur, 2021-02-10:
Created a new issue to address missing enum docs in other drivers - #1563. I'll close this issue.

closed 2021-02-10 by sbethur