ni/nimi-python #1557: Remove get_pattern_name from public API in nidigital

sbethur, 2021-02-04:
### Description of issue

I think this method is not required to be part of the public API since none of the other API accept/return pattern_index:
https://github.com/ni/nimi-python/blob/30687660fc0bc57b32d6b9639d95f55a52a7a312/generated/nidigital/nidigital/session.py#L1938

sbethur, 2021-02-05:
TL;DR: These methods can be private.

Consulted with NI-Digital API experts.. These methods are useful only in C API, where data returned by Fetch History RAM contains pattern indices and time set indices and users can use the methods in question to go from indices to names. But in Python (and in LabVIEW and .NET), History RAM data already contains the pattern and time set names. Hence these methods are not useful to clients. We must've have missed this during API review, but we can fix it while going to 1.0.

@ni-jxie / @phindman Can you please address this issue as part of nidigital 1.0 milestone?

FYI: @marcoskirsch

phindman, 2021-02-05:
The implementation of `fetch_history_ram_cycle_information`, as well as a few system tests, currently call these functions.  If we make them private, would it be acceptable to have those places call the private versions (i.e. `_get_pattern_name` and `_get_time_set_name`) instead?

phindman, 2021-02-05:
Just looking at the system tests, it seems like we could just delete the relevant test code.  The tests are testing the public functions, and they aren't public anymore.  But...
> in Python (and in LabVIEW and .NET), History RAM data already contains the pattern and time set names.

Taking a closer look at the [implementation of `fetch_history_ram_cycle_information`](https://github.com/ni/nimi-python/blob/master/src/nidigital/templates/session.py/fancy_fetch_history_ram_cycle_information.py.mako#L52), I don't think there is any way we can avoid calling these functions because that is how we get the names to put in the returned History RAM data.  So it might make sense to keep the tests too?

sbethur, 2021-02-09:
> I don't think there is any way we can avoid calling these functions because that is how we get the names to put in the returned History RAM data. 

Yes, we need to have these methods for `fetch_history_ram_cycle_information`. But the user doesn't have to call them. Hence we can make them private by adding `'codegen_method': 'private'` to their metadata. We will also need to update references to them in the hand-coded fancy implementation of `fetch_history_ram_cycle_information`.

Once we make them private, I don't think it's right for system tests to validate them. We either validate them using unit tests or indirectly via `fetch_history_ram_cycle_information`. In this case, I think it's sufficient to go with option 2 - Ensure the pattern and time set names returned by `fetch_history_ram_cycle_information` is correct.

Cc: @phindman

phindman, 2021-02-09:
OK, I'll remove them from the system tests then.

PR #1561 (2021-02-09): Make get_pattern_name and get_time_set_name private
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).

- [ ] ~I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~

- [ ] ~I've added tests applicable for this pull request~

### What does this Pull Request accomplish?

Makes get_pattern_name and get_time_set_name private, and fixes up references to those functions.

### List issues fixed by this Pull Request below, if any.

* Fix #1557 

### What testing has been done?

Just buil [... cut]
closed 2021-02-09 by marcoskirsch