ni/nimi-python #1584: Add "deprecated" functions to nimi-python's code generator

marcoskirsch, 2021-04-14:
### Description of issue

We need the ability to mark functions as "deprecated" in metadata so that the codegenerator does the appropriate thing:

* DeprecationWarning
* remove from ReadTheDocs or mark it as deprecated

This is needed for #1403

ni-jfitzger, 2025-01-08:
We should have the ability to deprecate enums, enum values and attributes, as well.

bkeryan, 2025-05-12:
In https://github.com/ni/nidaqmx-python and https://github.com/ni/measurement-plugin-python we have been using the https://pypi.org/project/deprecation/ package, which provides a decorator that generates a warning and adds a deprecated notice to the docstring.

However, Python 3.13 adds a standard decorator for this, [@warnings.deprecated](https://docs.python.org/3/library/warnings.html#warnings.deprecated), so I will not be using the `deprecation` package in new code. I will use the backport in https://github.com/python/typing_extensions/

Also, the docstring manipulation done by the `deprecation` package only works with doc tools that load your code at runtime, like autodoc, not doc tools that parse your code without loading it, like sphinx-autoapi.