ni/nimi-python #1602: Documentation for attribute repeated capabilities should be updated

luisgomes252, 2021-05-12:
### Description of issue

The example provided in the repeated capabilities documentation is the same for all kinds capabilities, and it's too specific to channels:

       .. code:: python

            session.channels['0-2'].channel_enabled = True

        passes a string of :python:`'0, 1, 2'` to the set attribute function.

1. '0-2' is not a goo example for instruments rep cap
2. "channel_enabled" makes no sense to be set on instruments
3. "passes a string of :python:'0, 1, 2' to the set attribute function" is irrelevant to Python programmers.

sbethur, 2021-06-10:
@luisgomes252, this can be closed, right?

luisgomes252, 2021-06-10:
> @luisgomes252, this can be closed, right?

It can. Sorry, that was an oversight on my part. I should have included it in [the PR](https://github.com/ni/nimi-python/pull/1600).

closed 2021-06-10 by sbethur