ni/nimi-python #1606: Table that lists characteristics of property has superfluous info

marcoskirsch, 2021-05-14:
Our documentation for properties contains:

```
        The following table lists the characteristics of this property.

            +-----------------------+------------+
            | Characteristic        | Value      |
            +=======================+============+
            | Datatype              | str        |
            +-----------------------+------------+
            | Permissions           | read-write |
            +-----------------------+------------+
            | Repeated Capabilities | None       |
            +-----------------------+------------+
            | Resettable            | No         |
            +-----------------------+------------+
```

* **Repeated Capabilities** is confusing on APIs that don't have them (`nitclk`, `nise`, `nimodinst` and to a lesser degree because it is based on an IVI driver `nidmm`
* **Resettable** is useless because while a small subset of IVI based NI drivers actually implement it, it is not exposed in `nimi-python` at all. See #531 

We should improve our generated documentation to remove **Resettable** and to only add **Repeated Capabilities** for drivers that have them.

luisgomes252, 2021-05-17:
@marcoskirsch, with respect to the "drivers that have them" part, would this be based on some metadata flag somewhere, or should we just hardcode this list somewhere in nimi-python? If the latter, right now I see repeated capabilities in dcpower, switch, digital and scope. Are there any others?

luisgomes252, 2021-05-17:
@marcoskirsch, with respect to Resettable, I'm seeing a lot of attributes with that attribute enabled in the internal metadata. For example, [most of the nidigital attributes](https://p4.natinst.com/browser/perforce/MI/Digital/PBD/SWCode/niDigitalAPIMetadata/master/source/niDigitalAPIMetadata/base/attributes/attributes.hapigen?revision_at=54700871#line_3), but also in other drivers, like fgen and scope. I assume you don't want to lose this information in the metadata. So are you proposing that we just ignore this attribute for now, for the purpose of doc generation? And will we put it back in when #531 gets implemented? If so, I need to add a comment to #531 to that effect.

marcoskirsch, 2021-05-18:
>@marcoskirsch, with respect to the "drivers that have them" part, would this be based on some metadata flag somewhere, or should we just hardcode this list somewhere in nimi-python? If the latter, right now I see repeated capabilities in dcpower, switch, digital and scope. Are there any others?

Global driver config metadata knows about repeated capabilities. See:
https://github.com/ni/nimi-python/blob/master/src/nidigital/metadata/config.py#L42

> @marcoskirsch, with respect to Resettable, I'm seeing a lot of attributes with that attribute enabled in the internal metadata. For example, [most of the nidigital [...]

Internal driver API metadata is not all relevant to nimi-python. The API metadata that we publish to this repo is nimi-python specific. Since the nimi-python APIs do not support resetting properties even when the underlying driver runtime does, there is no need to include that information in the nimi-python metadata nor is it beneficial to document it – it is actually counterproductive.

PR #1610 (2021-06-01): Instrument repeated capabilities doc improvements
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nimi-python/blob/master/CONTRIBUTING.md).
- [ ] ~~I've updated [CHANGELOG.md](https://github.com/ni/nimi-python/blob/master/CHANGELOG.md) if applicable.~~
- [x] I've added tests applicable for this pull request

### What does this Pull Request accomplish?

Improves attribute repeated capability documentation

### List issues fixed by this Pull Request below, if any.

* Fix #1604
* Fix #1606

### What testing has been done?

Local code generation w/ documentation inspecti [... cut]
closed 2021-06-01 by sbethur