ni/nimi-python #1617: Please check the Python code I wrote to control NI_Digital to read and write the "MIPI" register

Super-leilei, 2021-05-31:
I am using "Pxie 6570" to read and write" MIPI "register. I am using the" nidigital "library in Python, but my register read back is always 0. Please help me check the code, thank you very much!
Attached are the Python code and the project file of the “Digital Pattern”
[python_code.zip](https://github.com/ni/nimi-python/files/6568759/python_code.zip)
 @marcoskirsch @lkolbly @juno-jin-ni

smooresni, 2021-06-01:
@Super-leilei you might find [this](https://github.com/NISystemsEngineering/mipi-rffe-python/tree/oop-dev) reference architecture to be a good starting point.

sbethur, 2021-06-01:
@Super-leilei, please use GitHub issues only for ideas, enhancements, tasks, or bugs in nimi-python.

closed 2021-06-01 by sbethur