ni/nimi-python #1627: nifgen API reference does not mention which properties and methods can be called on a repeated capabilities container

sbethur, 2021-06-11:
### Description of issue

According to the documentation, [these ](https://nimi-python.readthedocs.io/en/1.3.3/nifgen/rep_caps.html#)repeated capabilities are supported in nifgen. But the API reference for properties and methods doesn't mention which of them can be called on a repeated capabilities container.

sbethur, 2022-03-28:
@ni-jfitzger, I remember you looking into this issue; can you please provide the latest status on this?

ni-jfitzger, 2022-03-28:
The PR lined above addressed this issue for properties, but not methods.
But looking at [the latest documentation](https://nimi-python.readthedocs.io/en/master/nifgen/class.html#methods), several methods have tips that mention they can be called on specific channels and show the syntax. I'm not sure if we have any methods that support any repeated capabilities other than channels. Maybe this can be closed?

marcoskirsch, 2023-03-23:
Maybe not in `nifgen`, but in `nidigital` we do have functions that take in pin names.