ni/nimi-python #1628: Interpreter exits with code -1073740791 (0xC0000409) if simulated nidcpower session isn't closed

smooresni, 2021-06-16:
### Description of issue

If a simulated nidcpower session is opened and allowed to remain open when the interpreter exits, a crash with exit code -1073740791 (0xC0000409).

### System report

`python -c "import niscope; niscope.print_diagnostic_information()"` output (replace `niscope` with appropriate package name)

    ```
OS:
    Name:      Windows
    Version:   10.0.17763
    Bits:      64
Driver:
    Name:      NI-DCPower
    Version:   20.7.0.49576
Module:
    Name:      nidcpower
    Version:   1.3.3
Python:
    Version:   3.6.8 (tags/v3.6.8:3c6b436a57, Dec 24 2018, 00:16:47) [MSC v.1916 64 bit (AMD64)]
    Bits:      64
    Is_Venv:   True
    Installed Packages:
        setuptools==40.6.2
        pywin32==301
        pip==21.1.2
        nitsm==0.0.1
        nidcpower==1.3.3
        hightime==0.2.0

    ```

### Steps to reproduce issue

1. In a new python file, place the line:

    ``` python
  nidcpower.Session("Dev1", options={"Simulate": True, "DriverSetup": {"Model": "4162"}})
    ```

2. Run and observe exit code.

marcoskirsch, 2022-03-28:
This is most likely a driver runtime, some unjoined thread of something like that. Needs further investigation by driver engineer.=s.

marcoskirsch, 2022-07-21:
Closing as this is not a nimi-python bug.
Please close sessions prior exiting. I believe using `with` statement will take care of it for you.

closed 2022-07-21 by marcoskirsch