ni/nimi-python #1643: Method bug in NIDigital: write_source_waveform_broadcast

Super-leilei, 2021-07-19:
@sbethur
[image]

I found a bug in the method "write_source_waveform_broadcast" when using NIDigital library. Every time I call this method, if the number of bytes is different from the number of bytes written in the first time, it will report an error. If I write the second time, I cannot close VIO. I want to keep the state of the instrument when I wrote the first time. The screenshot of the error report is as follows. Please help to check the error, thank you!
[image]

sbethur, 2021-07-19:
@Super-leilei, The "Specified waveform  has already been written with a different waveform size" error when the number of bytes is different is expected behavior in the driver runtime.

>  If I write the second time, I cannot close VIO.

Can you please post the code that shows this issue?

Super-leilei, 2021-07-21:
@sbethur
The complete code is as follows. When running to "a.extended_register_write(0x8,0x1B,[0x38,0x38])", the above error will be reported. Please help to check the reason, thank you!
[bug.zip](https://github.com/ni/nimi-python/files/6853649/bug.zip)

Super-leilei, 2021-07-26:
@sbethur Hi,Sbethur  May I ask if there is a solution to this problem? Does anyone follow up this problem? Because the project is urgent, I hope to get your reply as soon as possible, thank you!

Super-leilei, 2021-08-05:
@sbethur HI,Is there any progress on this issue recently? Thank you!

sbethur, 2021-08-13:
@super-leilei, based on your code, it looks like the `waveform_data` passed to the 2nd call to `write_source_waveform_broadcast` is longer than in the 1st call. Calling `write_source_waveform_broadcast` with different sizes is not a supported use case and hence the error you're seeing above is expected behavior.

closed 2021-08-13 by sbethur

Super-leilei, 2021-08-16:
@sbethur Hi sbethur ,In practical application, we need the second time to write waveform length is different from the first time, in this case, what method can be used to do it ?Thank you!

marcoskirsch, 2021-08-16:
Hi @Super-leilei,

The Issues section of the nimi-python project is meant for tracking bugs related to the Python APIs for NI’s Modular Instruments. This issue is not a bug on either the `nidigital` Python module nor the underlying NI-Digital Pattern Driver runtime.

There are a few NI-Digital Pattern Driver usage experts roaming around here, and they will help if they can, but the right place and the more effective one to ask for help using the products is in [NI’s Discussion Forums](https://forums.ni.com/t5/Digital-I-O/bd-p/70?profile.language=en). Please consider posting there if you haven’t.

Good luck!

Super-leilei, 2021-08-17:
@marcoskirsch Hi  marcoskirsch,Thank you for your recommendation,I did not find any recommended use of this method in the "NI" forum, and contacted your expert, who also said that this method is not suitable for practical use, the "write_source_waveform_broadcast" method is fine, but in practical use, Not sure every time write wave are the same size, may be the first time to write the waveform size is one byte, second write two bytes, so if this method does not support the actual application, this is not a BUG, or where the method was invented meaning, finally also please @sbethur look at the problem, Or be to use what method more appropriate! Thank you! Thank you very much!

sbethur, 2021-08-17:
(Deleted a duplicate comment)

@Super-leilei, `write_source_waveform_broadcast` is currently designed to work with "same-sized" waveforms. If you want to source multiple waveforms of different sizes, you can always create them by calling `create_source_waveform_serial` multiple times with different names and then calling `write_source_waveform_broadcast` for each of them.

If your use-case doesn't work with different waveforms and you need to be able to change the size of a given waveform, please submit a feature request in NI's Discussion Forums mentioned above.