ni/nimi-python #1663: Session creation fails with Windows Error 0xc06d007f

sbethur, 2021-10-01:
I'm running the nifgen_standard_function.py example and am unsure as to why it's producing this output:

```
File "C:\Users\att\Documents\nimi-python-master\nimi-python-master\src\nifgen\examples\nifgen_standard_function.py", line 47, in <module>
    main()
  File "C:\Users\att\Documents\nimi-python-master\nimi-python-master\src\nifgen\examples\nifgen_standard_function.py", line 33, in main
    _main(sys.argv[1:])
  File "C:\Users\att\Documents\nimi-python-master\nimi-python-master\src\nifgen\examples\nifgen_standard_function.py", line 29, in _main
    example(args.resource_name, args.option_string, args.waveform, args.frequency, args.amplitude, args.offset, args.phase, args.time)
  File "C:\Users\att\Documents\nimi-python-master\nimi-python-master\src\nifgen\examples\nifgen_standard_function.py", line 10, in example
    with nifgen.Session(resource_name=resource_name, options=options) as session:
  File "C:\Users\att\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\nifgen\session.py", line 3067, in __init__
    self._vi = self._initialize_with_channels(resource_name, channel_name, reset_device, options)
  File "C:\Users\att\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\nifgen\session.py", line 4091, in _initialize_with_channels
    error_code = self._library.niFgen_InitializeWithChannels(resource_name_ctype, channel_name_ctype, reset_device_ctype, option_string_ctype, None if vi_ctype is None else (ctypes.pointer(vi_ctype)))
  File "C:\Users\att\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\nifgen\_library.py", line 444, in niFgen_InitializeWithChannels
    return self.niFgen_InitializeWithChannels_cfunc(resource_name, channel_name, reset_device, option_string, vi)
OSError: [WinError -1066598273] Windows Error 0xc06d007f
```

I have downloaded the NI-FGEN (version 20.0.0) driver from the National Instruments website and am using python module nifgen=1.3.2.
Any tips on how to fix the error would be appreciated!

_Originally posted by @emtagg in https://github.com/ni/nimi-python/discussions/1536_

marcoskirsch, 2022-01-06:
This is tracked internally at NI by AB#1659694

sbethur, 2022-04-25:
Update on internal bug (AB#1659694): @dylan-matthew-turner did some good debugging and root-caused the issue.

**Summary:** The failure happens when any nimi-python session is opened using the Python interpreter available in the Microsoft store.

**Details:** Python available in the Microsoft store is an [UWP app](https://docs.microsoft.com/en-us/windows/uwp/get-started/universal-application-platform-guide), which has stricter restrictions on loading dependent libraries. Some dependent libraries do not comply with these requirements, and hence session creation fails.

**Workaround:** Use Python available elsewhere like [Python.org](https://www.python.org/downloads/).

closed 2022-07-21 by marcoskirsch