ni/nimi-python #1680: System tests are getting automatically triggered for pull requests

sbethur, 2022-01-10:
### Description of issue

System tests are getting automatically triggered for pull requests, if the PR is from a branch in `ni/nimi-python` repo. If the PR is coming from a fork, then the system tests are getting triggered as expected.

SukruthGrandhiNI, 2022-01-14:
I'm unable to reproduce the issue.

The pipeline got triggered immediately after raising the PR(NI account) from a branch.
PR [link](https://github.com/ni/nimi-python/pull/1683).
Azure Pipelines [link](https://ni.visualstudio.com/DevCentral/_build/results?buildId=2013809&view=results)

The pipeline got triggered only after commenting(/Azurepipelines run) for the PR(external contributor) from a branch.
PR [link](https://github.com/ni/nimi-python/pull/1684)
Azure pipelines [link](https://ni.visualstudio.com/DevCentral/_build/results?buildId=2014540&view=results)

SukruthGrandhiNI, 2022-01-14:
@sbethur can you close the issue if it is not reproducible?

sbethur, 2022-01-25:
> @sbethur can you close the issue if it is not reproducible?

Closing since the auto-trigger is working fine now.

closed 2022-01-25 by sbethur