ni/nimi-python #1690: Cannot run nidigital unit tests on Windows without NI-TClk driver runtime installed

sbethur, 2022-01-31:
### Description of issue

nimi-python unit tests should not need driver runtimes to be installed. This is true if they are run on Linux or on WSL in Windows. But when run on Windows, an error is returned:
```
generated\nidigital\nidigital\unit_tests\test_nidigital.py:102: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
generated\nidigital\nidigital\session.py:3777: in __init__
    self.tclk = nitclk.SessionReference(self._vi)
.tox\64\py36-test\lib\site-packages\nitclk\session.py:161: in __init__
    self._library = _library_singleton.get()
.tox\64\py36-test\lib\site-packages\nitclk\_library_singleton.py:50: in get
    ctypes_library = ctypes.CDLL(_get_library_name())
```

sbethur, 2022-02-01:
We're running into a regression introduced in Python 3.8: https://bugs.python.org/issue39243

`None` is returned by `_get_library_name()` when NI-TClk driver runtime isn't installed and `ctypes.CDLL` constructor cannot deal with `None` in Python 3.8 and 3.9.

dylan-matthew-turner, 2022-04-08:
This is a problem with the ctypes library which affects unit tests, so it's not visible to users, and it only fails on Windows, so it does not affect contributors.
We don't actually call to the underlying library in the unit tests, so we could fix this by creating a mock ctypes library as a workaround. However, considering the small coverage and the fact that we'll have to undo the workaround once the bug is fixed, I recommend closing this due to low ROI.

closed 2022-04-11 by sbethur