ni/nimi-python #1698: Returned parameter which is marked as unused in Python API  still gets codegen-ed into the documentation

sbethur, 2022-03-17:
There is an existing bug with the codegen logic where return parameter which is marked as unused in Python API (i.e. by specifying 'use_in_python_api': False in python/functions/functions.hapigen) still gets codegen-ed into the documentation
[image]
although it no longer gets returned by the actual code:

return [ni_lcr_measurement.NILCRMeasurement(measurements_ctype[i]) for i in range(count_ctype.value)]

Include relevant system information, steps to reproduce, and any workarounds.

ni-jfitzger, 2022-09-07:
My guess: change this line: https://github.com/ni/nimi-python/blob/f3324c733daaa52f7d7d2cc6a131d2268f77404d/build/helper/metadata_filters.py#L251

to
`        if x['direction'] == 'out' and (options_to_use['skip_output_parameters'] or not x['use_in_python_api']):`

ni-jfitzger, 2023-03-23:
Fixed by #1822

ni-jfitzger, 2023-03-23:
Fixed by #1822

closed 2023-03-23 by ni-jfitzger