ni/nimi-python #1774: NI-DCPower: Do not support mixing of fully-qualified channel names (FQCNs) and non-FQCNs in repeated capabilities string

olsl21, 2022-07-04:
### Description of issue

Currently, the NI-DCPower driver runtime supports passing in a repeated capabilities string that mixes fully-qualified channel name(s) (i.e. channel name(s) with instrument prefix) and non-fully-qualified channel name(s). Examples of such repeated capabilities string include "4,Dev1/1:2". 

The desired behavior is not to support mixing FQCNs and non-FQCNs.

### System report

`python -c "import nidcpower; nidcpower.print_diagnostic_information()"` output:

```
OS:
    Name:      Windows
    Version:   10.0.18362
    Bits:      64
Driver:
    Name:      NI-DCPower
    Version:   22.5.0
Module:
    Name:      nidcpower
    Version:   1.4.1
Python:
    Version:   3.7.9 (tags/v3.7.9:13c94747c7, Aug 17 2020, 18:58:18) [MSC v.1900 64 bit (AMD64)]
    Bits:      64
    Is_Venv:   True
    Installed Packages:
        setuptools==47.1.0
        pip==20.1.1
        nidcpower==1.4.1
        hightime==0.2.1
 ```

### Steps to reproduce issue

``` python
import nidcpower

with nidcpower.Session(resource_name="Dev1", channels="", options="Simulate=1, DriverSetup=Model:4162; BoardType:PXIe") as smu:
    with smu.initiate():
        smu.channels["4,Dev1/1:2"].measure_multiple()
```

marcoskirsch, 2022-07-21:
@olsl21 I think this issue is incorrect. Isn't it right that NI-DCPower session with independent channels _does support_ channel strings such as "4,Dev1/1:2" if and only if only 1 instrument is involved in the session?

If so, we can simply close.

olsl21, 2022-07-22:
@marcoskirsch Yes, NI-DCPower session with independent channels does support channel strings such as "4,Dev1/1:2" if and only if only 1 instrument is involved in the session. This issue was raised to _disable_ this support, but I couldn't trace back to the associated PR comment that mentioned about this.

Since you're ok with this behavior, I'll close this issue.

closed 2022-07-22 by olsl21