ni/nimi-python #1807: nifgen Python API streaming support ?

okz, 2022-08-11:
The Python nifgen API seems to have streaming support, but haven't seen any examples or tests using streaming. Are there any known limitations ? 

Planning to test following "\niFgen\documentation\English\SigGenHelp.chm::/streaming.html"  but wondered if the indicated transfer rates can be achieved with the Python API.  I am hoping ~50MB/s transfer rates would be possible. 

[image]

Kind regards,
Oz

marcoskirsch, 2022-08-17:
You will be limited by system performance. I'm curious to see if you can accomplish the desired rate on your setup. Please report back!

(I expect you will be able to hit it, but I don't know what PC you have nor what instrument model you are using).

okz, 2022-12-04:
Just to update that, using numpy arrays and using the python API 's lower level private _write functions, ~40MB/s was achieved. 

Best results were through various remote controllers, MXI-3 etc, using modern CPU's. The NI embedded controllers I had access to were 15years old, and didn't get much above 20MB/s

marcoskirsch, 2022-12-05:
Good to hear. I am a bit baffled that you had to use "lower level private _write functions". Those are a bad idea for your code to call into directly as we make no source compatibility guarantees for them. Did using the public methods impact your code's performance severly?

marcoskirsch, 2022-12-05:
Consider pasting a small example of how you accomplished it.

closed 2022-12-15 by marcoskirsch