ni/nimi-python #1812: NI-DCPower: cannot establish a session. Error "function 'niDCPower_InitializeWithIndependentChannels' not found"

dbbotkin, 2022-08-26:
### Description of issue

    <Using nidcpower to access PXI-4132 on PXI-1002 running Win7 SP1 with PXI-8105 controller, cannot establish a session due to error: "function 'niDCPower_InitializeWithIndependentChannels' not found"
Note: same system able to interact with nidmm and nifgen on PXI-4070 and PXI-5404 modules, no problem.>

### System report

`python -c "import niscope; niscope.print_diagnostic_information()"` output (replace `niscope` with appropriate package name)

    ```
   OS:
    Name:      Windows
    Version:   6.1.7601
    Bits:      32
Driver:
    Name:      NI-DCPower
    Version:   17.6.1.49153
Module:
    Name:      nidcpower
    Version:   1.4.2
Python:
    Version:   3.7.9 (tags/v3.7.9:13c94747c7, Aug 17 2020, 18:01:55) [MSC v.1900 32 bit (Intel)]
    Bits:      32
    Is_Venv:   False
    Installed Packages:
        nitclk==1.4.2
        nifgen==1.4.2
        nidmm==1.4.2
        nidcpower==1.4.2
        hightime==0.2.1
        enum34==1.1.10
        wrapt==1.12.1
        wheel==0.36.2
        websockets==9.1
        typing-extensions==3.10.0.0
        typed-ast==1.4.3
        toml==0.10.2
        thonny==3.3.13
        six==1.16.0
        setuptools==47.1.0
        send2trash==1.7.1
        reedsolo==1.5.4
        pyserial==3.5
        pynacl==1.4.0
        pylint==2.9.5
        pycparser==2.20
        ptyprocess==0.7.0
        pip==20.1.1
        parso==0.8.2
        paramiko==2.7.2
        mypy==0.910
        mypy-extensions==0.4.3
        mccabe==0.6.1
        lazy-object-proxy==1.6.0
        jedi==0.18.0
        isort==5.9.2
        esptool==3.1
        ecdsa==0.17.0
        docutils==0.17.1
        cryptography==3.4.7
        colorama==0.4.4
        cffi==1.14.6
        bitstring==3.1.9
        bcrypt==3.2.0
        asttokens==2.0.5
        astroid==2.6.5
    ```

### Steps to reproduce issue

    ``` python
    <
import nidcpower
#resource_name, channels=None, reset=False, options={}, independent_channels=True
options = { 'driver_setup': {'Model': '4132', 'BoardType': 'PXI', }, }
PXI4132 = nidcpower.Session("PXI-4132", channels='0', reset=True, options=options, independent_channels=True)
PXI4132.close()>
    ```
>>> %Run PXI-4132Volts.py
Traceback (most recent call last):
  File "C:\Users\Donal\AppData\Roaming\Python\Python37\site-packages\nidcpower\_library.py", line 97, in _get_library_function
    function = getattr(self._library, name)
  File "C:\Users\Donal\AppData\Local\Programs\Thonny\lib\ctypes\__init__.py", line 378, in __getattr__
    func = self.__getitem__(name)
  File "C:\Users\Donal\AppData\Local\Programs\Thonny\lib\ctypes\__init__.py", line 383, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'niDCPower_InitializeWithIndependentChannels' not found

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "P:\Ni-DMM\PXI-4132Volts.py", line 4, in <module>
    PXI4132 = nidcpower.Session("PXI-4132", channels='0', reset=True, options=options, independent_channels=True)
  File "C:\Users\Donal\AppData\Roaming\Python\Python37\site-packages\nidcpower\session.py", line 7441, in __init__
    self._vi = self._fancy_initialize(resource_name, channels, reset, options, independent_channels)
  File "C:\Users\Donal\AppData\Roaming\Python\Python37\site-packages\nidcpower\session.py", line 7765, in _fancy_initialize
    return self._initialize_with_independent_channels(resource_string, reset, option_string)
  File "C:\Users\Donal\AppData\Roaming\Python\Python37\site-packages\nidcpower\session.py", line 8243, in _initialize_with_independent_channels
    error_code = self._library.niDCPower_InitializeWithIndependentChannels(resource_name_ctype, reset_ctype, option_string_ctype, None if vi_ctype is None else (ctypes.pointer(vi_ctype)))
  File "C:\Users\Donal\AppData\Roaming\Python\Python37\site-packages\nidcpower\_library.py", line 377, in niDCPower_InitializeWithIndependentChannels
    self.niDCPower_InitializeWithIndependentChannels_cfunc = self._get_library_function('niDCPower_InitializeWithIndependentChannels')
  File "C:\Users\Donal\AppData\Roaming\Python\Python37\site-packages\nidcpower\_library.py", line 99, in _get_library_function
    raise errors.DriverTooOldError() from e
nidcpower.errors.DriverTooOldError: A function was not found in the NI-DCPower runtime. Please visit http://www.ni.com/downloads/drivers/ to download a newer version and install it.
>>>

marcoskirsch, 2022-08-27:
The exception says it right there:

```
nidcpower.errors.DriverTooOldError: A function was not found in the NI-DCPower runtime. Please visit http://www.ni.com/downloads/drivers/ to download a newer version and install it.
```

Alternatively, you can pass `independent_channels=True` to `nidcpower.Session`’s constructor to get the old behavior. But I suggest you use the latest software whenever possible 🙂.

closed 2022-08-27 by marcoskirsch

dbbotkin, 2022-08-27:
Hello Marcos,
As I posted, I am running on an older system with Windows 7 and have been working to fix this for several days. Of course I tried to use a “newer” version—and also an “older” one with much the same result. NI has moved to Windows 10+ and I think that is a mistake IMHO, at least until Microsoft gets rid of the bloatware in Win 10. In any case, the highest version I can use is 18 (tested that) but switched back to 17.6 as it was the version listed as “tested-on” in the release. 

I realize there is more to the interface with a SMU than with a DMM or FGEN, and probably the complexity is there to support current models.   Is it possible to get the run-time dcpower without having to upgrade the rest? 

Thanks for your contribution to getting Python working at NI.

Cheers,
Donal

marcoskirsch, 2022-08-28:
> As I posted, I am running on an older system with Windows 7

:(

If you're using Windows 7 then you will have to use an older runtime. It's not as simple as you think for NI to keep supporting Windows 7 – not only due to the added testing burden but because the kernel binary signing facilities from Microsoft no longer exist.

You should be able to use the alternative of passing  `independent_channels=False`. Let me know if that doesn't work for you.

dbbotkin, 2022-08-28:
"It's not as simple as you think for NI to keep supporting Windows 7”
I don’t think it is simple at all to support Gen 1 when Gen 3 is shipping and Gen 5 is what everyone is working on. (It might be different at NI) I am pretty sure MSFT is in that loop and will be for a long time. That is the model for latest-is-better technology companies, however there are other models. There are companies that can provide parts and service for items sold decades back—mostly in Europe—even some in the USA. I think customers expect more from hardware vendors and are willing to settle for whatever technology is bright and shiny and put the old tech in the dumpster.

Right now MSFT is doing all it can to kill Win 7: example, the vigor with which the software is claiming “This is not genuine” so the users will upgrade. OK, most users should as Win 7 is an unlocked car with the window down and a backpack on the front seat. Not happy are a horde of small businesses running proprietary software on a dusty PC under someone’s desk or a thousand POS terminals that simply cannot be upgraded. This is a nightmare for Joe at the job-shop and probably not a lot of joy in the MSFT legal department either.

Some companies would make the software open-source in a final version and ‘donate it’ to a NFP with some old engineers that worked on it back-when to build a users’ group and answer questions. That will not happen at MSFT.

My open question on Win 10 is whether a version without the bloatware might possibly run on the hardware I have.  BTW, I started with Windows 3.1 then 98, ME, XP, 7 and finally 10. I do not like the UI of 10 one bit.

Cheers,
Donal