ni/nimi-python #1820: windows-1251 is hardcoded in our APIs as the encoding to use

ni-jfitzger, 2022-09-19:
### Description of issue

    'windows-1251' is hardcoded in our APIs as the encoding to use. We support Linux, as well, so this definitely doesn't seem right.

### System report
None

### Steps to reproduce issue

Just look at the code.

ni-jfitzger, 2022-09-19:
Unfortunately, @marcoskirsch, if we discussed what the correct solution is, I've already forgotten it.

marcoskirsch, 2022-09-19:
We need to figure out what the correct encoding in Linux is for our driver runtimes, then select the right encoding based on the platform we're running on.

I think a good way to test is to use two devices whose aliases have non-ascii characters, and confirm we can create sessions and make routes between them.

marcoskirsch, 2022-09-19:
Once we get this working right, we should ensure there is unit-test coverage.

bkeryan, 2023-03-02:
Note that `windows-1251` is for languages that use Cyrillic script, such as Russian, Ukrainian, and Belarusian. You probably meant to hardcode `windows-1252`, which is roughly equivalent to Latin-1.