ni/nimi-python #1894: `nidcpower.OutputCutoffReason` is missing enum values

marcoskirsch, 2023-03-10:
### Description of issue

NI-DCPower 23.0 released. The header file lists the following possible values for functions `niDCPower_QueryLatchedOutputCutoffState()` and `niDCPower_ClearLatchedOutputCutoffState()`"

```C
/*- Defined values for OutputCutoffReason values -*/
--
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_ALL                                           (-1L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_VOLTAGE_OUTPUT_HIGH                           (1L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_VOLTAGE_OUTPUT_LOW                            (2L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_CURRENT_MEASURE_HIGH                          (4L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_CURRENT_MEASURE_LOW                           (8L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_VOLTAGE_CHANGE_HIGH                           (16L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_VOLTAGE_CHANGE_LOW                            (32L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_CURRENT_CHANGE_HIGH                           (64L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_CURRENT_CHANGE_LOW                            (128L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_VOLTAGE_SATURATED                             (256L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_CURRENT_SATURATED                             (512L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_VOLTAGE_MEASURE_HIGH                          (1024L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_VOLTAGE_MEASURE_LOW                           (2048L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_SELF_TEST_MEASUREMENT_HIGH                    (4096L)
#define NIDCPOWER_VAL_OUTPUT_CUTOFF_REASON_SELF_TEST_MEASUREMENT_LOW                     (8192L)
```

However, the [latest `nidcpower.OutputCutoffReason`](https://github.com/ni/nimi-python/blob/2dfa5c3cfc65b6ea594d58214a5406ae06d3a35e/generated/nidcpower/nidcpower/enums.py#L373) lists:

```
class OutputCutoffReason(Enum):
    ALL = -1
    r'''
    Queries any output cutoff condition; clears all output cutoff conditions.
    '''
    VOLTAGE_OUTPUT_HIGH = 1
    r'''
    Queries or clears cutoff conditions when the output exceeded the high cutoff limit for voltage output.
    '''
    VOLTAGE_OUTPUT_LOW = 2
    r'''
    Queries or clears cutoff conditions when the output fell below the low cutoff limit for voltage output.
    '''
    CURRENT_MEASURE_HIGH = 4
    r'''
    Queries or clears cutoff conditions when the measured current exceeded the high cutoff limit for current output.
    '''
    CURRENT_MEASURE_LOW = 8
    r'''
    Queries or clears cutoff conditions when the measured current fell below the low cutoff limit for current output.
    '''
    VOLTAGE_CHANGE_HIGH = 16
    r'''
    Queries or clears cutoff conditions when the voltage slew rate increased beyond the positive change cutoff for voltage output.
    '''
    VOLTAGE_CHANGE_LOW = 32
    r'''
    Queries or clears cutoff conditions when the voltage slew rate decreased beyond the negative change cutoff for voltage output.
    '''
    CURRENT_CHANGE_HIGH = 64
    r'''
    Queries or clears cutoff conditions when the current slew rate increased beyond the positive change cutoff for current output.
    '''
    CURRENT_CHANGE_LOW = 128
    r'''
    Queries or clears cutoff conditions when the current slew rate decreased beyond the negative change cutoff for current output.
    '''
```

Several are missing, so the functionality is not available to customers.

ni-jfitzger, 2023-03-13:
It turns out these missing values aren't intended for real usage and have no public documentation. I'd prefer we not add these undocumented values unless we get a request to do so.

ni-jfitzger, 2023-03-28:
After some internal searching, we determined (based on Pull Request discussions and the Feature Spec) that these extra values were never intended to be publicly supported, though they were somehow released in the public header, anyway.

We will not be adding them to the Python API.

closed 2023-03-28 by ni-jfitzger