ni/nimi-python #1904: [DCPower] Error 0xc06d007f on session open

kiler129, 2023-03-30:
### Description of issue
When trying to open a new `nidcpower` session I'm getting a `OSError: [WinError -1066598273] Windows Error 0xc06d007f` exception on `niDCPower_InitializeWithIndependentChannels_cfunc`. I'm running locally on a PXIe-1085 chassis with an SMU (4141) in slot 13.

Googling around I only came up with this being a generic Win32 runtime error. I installed NI-DCPower driver (2023 Q1) as well as PXI Platform Services (2023 Q1).  The SMU is visible and passes self-test from NI MAX as well. What's the most bizarre the session opens without a problem from a different user account on the same machine.

### System report
NI DCPower:

```
OS:
    Name:      Windows
    Version:   10.0.19044
    Bits:      64
Driver:
    Name:      NI-DCPower
    Version:   23.0.0
Module:
    Name:      nidcpower
    Version:   1.4.3
Python:
    Version:   3.10.10 (tags/v3.10.10:aad5f6a, Feb  7 2023, 17:20:36) [MSC v.1929 64 bit (AMD64)]
    Bits:      64
    Is_Venv:   True
    Installed Packages:
        setuptools==65.5.0
        pip==23.0.1
        pyyaml==6.0
        pyserial==3.5
        numpy==1.24.2
        nidcpower==1.4.3
        iso8601==1.1.0
        future==0.18.3
        hightime==0.2.1
        serial==0.0.97
```

### Steps to reproduce issue

```python
import nidcpower

nidcpower.Session(resource_name="PXI1Slot13") # I also tried adding /0 to no avail
```

marcoskirsch, 2023-03-31:
Likely a duplicate of #1663
Please confirm where you got Python from.

kiler129, 2023-03-31:
> Likely a duplicate of #1663
> Please confirm where you got Python from.

You're 100% correct - that was the issue. Is there some clever way to detect MS Store python? The sys.platform doesn't report it as UWP.
It's a problematic issue as typing "python" when not installed automatically suggest MS Store python install now.

closed 2023-03-31 by kiler129