ni/nimi-python #1959: `niswitch.Session` doesn't support specifying an `options` dictionary or driver setup string

bkeryan, 2023-05-02:
### Description of issue

`niswitch.Session` doesn't support specifying an `options` dictionary like the other drivers. Instead, it has parameters for `simulate` and `topology`. 

Impact:
- Enabling session-based simulation requires different code than for the other drivers. (Note: if it supported `options`, you would still have to specify the `topology` key instead of the `model` key.)
- Specifying options other than `simulate` or driver setup keys other than `topology` is unsupported.

### System report

`python -c "import niswitch; niswitch.print_diagnostic_information()"` output

```
OS:
    Name:      Windows
    Version:   10.0.19045
    Bits:      64
Driver:
    Name:      NI-SWITCH
    Version:   22.8.0.49295
Module:
    Name:      niswitch
    Version:   1.4.4
Python:
    Version:   3.9.13 (tags/v3.9.13:6de2ca5, May 17 2022, 16:36:42) [MSC v.1929 64 bit (AMD64)]
    Bits:      64
    Is_Venv:   True
    Installed Packages:
        click==8.1.3
        colorama==0.4.6
        grpcio==1.54.0
        hightime==0.2.1
        mypy==1.2.0
        mypy-extensions==1.0.0
        ni-measurementlink-service==1.0.1
        niswitch==1.4.4
        pip==23.0.1
        protobuf==4.22.3
        pywin32==303
        setuptools==67.6.1
        tomli==2.0.1
        typing-extensions==4.5.0
        wheel==0.40.0
```

### Steps to reproduce issue

``` python
>>> import niswitch
>>> session = niswitch.Session(resource_name="", options={"simulate": True, "driver_setup": {"topology": "2567/Independent"}})
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: __init__() got an unexpected keyword argument 'options'
```

bkeryan, 2023-06-26:
@ni-jfitzger Are you referring to the .NET API? C and LV have `niSwitch_InitWithOptions` which supports specifying the topology as well as other options.

ni-jfitzger, 2023-06-26:
Sorry, I meant this part, specifically, is not unique to the Python API.

>Note: if it supported options, you would still have to specify the topology key instead of the model key.)