ni/nimi-python #1982: Broken link on the nidcpower documentation page

dimapu, 2023-06-14:
### Description of issue

This page (https://nidcpower.readthedocs.io/en/latest/class.html#reset) has a broken link, "[Programming States](https://nidcpower.readthedocs.io/en/latest/REPLACE_DRIVER_SPECIFIC_URL_1(programmingstates))" on the screenshot below.

[image]

### System report

NA

### Steps to reproduce issue

    1. Go to https://nidcpower.readthedocs.io/en/latest/class.html#reset
    2. Click on the link "Programming states"

ni-jfitzger, 2023-06-14:
Woohoo! People are looking at the new doc projects.

The broken link is "https://nidcpower.readthedocs.io/en/latest/REPLACE_DRIVER_SPECIFIC_URL_1(programmingstates)".
This seems very familiar. It may be a known issue.

ni-jfitzger, 2023-06-14:
Duplicate of #1487

dimapu, 2023-06-14:
I will close this then.

dimapu, 2023-06-14:
Thanks for your efforts!

closed 2023-06-14 by dimapu