ni/nimi-python #2001: Mixing an nidmm PXI-407x session and niswitch sessions causes unexpected failures finding the NI-SWITCH Functions on Linux

mgallatin, 2023-08-24:
### Description of issue

Using Python 3.9.15, NI-DMM 23.5 and the nidmm 1.4.5 python package, I get weird behavior when using a 407x device and mixing usage of NI-DMM and NI-SWITCH sessions (used the niswitch 1.4.5 package, but I also reproduced with the nidcpower 1.4.5 package).

Specifically after running a very small program (listed below in the steps to reproduce) that includes opening an NI-DMM session, calling dmm.initiate(), and then opening an NI-SWITCH Session, I get the following error:

```
Traceback (most recent call last):
  File "/home/user/.pyenv/versions/3.9.15/lib/python3.9/site-packages/niswitch/_library.py", line 65, in _get_library_function
    function = getattr(self._library, name)
  File "/home/user/.pyenv/versions/3.9.15/lib/python3.9/ctypes/__init__.py", line 387, in __getattr__
    func = self.__getitem__(name)
  File "/home/user/.pyenv/versions/3.9.15/lib/python3.9/ctypes/__init__.py", line 392, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /home/user/.pyenv/versions/3.9.15/bin/python: undefined symbol: niSwitch_InitWithTopology

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/user/./switch_reproducer.py", line 6, in <module>
    with niswitch.Session(resource_name="", topology="2535/1-Wire 4x136 Matrix", simulate=True) as session1:
  File "/home/user/.pyenv/versions/3.9.15/lib/python3.9/site-packages/niswitch/session.py", line 1377, in __init__
    self._interpreter.set_session_handle(self._init_with_topology(resource_name, topology, simulate, reset_device))
  File "/home/user/.pyenv/versions/3.9.15/lib/python3.9/site-packages/niswitch/session.py", line 1993, in _init_with_topology
    vi = self._interpreter.init_with_topology(resource_name, topology, simulate, reset_device)
  File "/home/user/.pyenv/versions/3.9.15/lib/python3.9/site-packages/niswitch/_library_interpreter.py", line 293, in init_with_topology
    error_code = self._library.niSwitch_InitWithTopology(resource_name_ctype, topology_ctype, simulate_ctype, reset_device_ctype, None if vi_ctype is None else (ctypes.pointer(vi_ctype)))
  File "/home/user/.pyenv/versions/3.9.15/lib/python3.9/site-packages/niswitch/_library.py", line 225, in niSwitch_InitWithTopology
    self.niSwitch_InitWithTopology_cfunc = self._get_library_function('niSwitch_InitWithTopology')
  File "/home/user/.pyenv/versions/3.9.15/lib/python3.9/site-packages/niswitch/_library.py", line 67, in _get_library_function
    raise errors.DriverTooOldError() from e
niswitch.errors.DriverTooOldError: A function was not found in the NI-SWITCH runtime. Please visit http://www.ni.com/downloads/drivers/ to download a newer version and install it.
```

The error is clearly misleading.  I shouldn't have any issues opening a session to both devices.

I am using IVI simulated devices, so no hardware is required.  
If I simulate a 4081 DMM instead of a 4071, the error doesn't happen.  
If I remove the dmm.initiate() call then the error doesn't happen.  
If I use Python 3.10.8 then even the reproducing example as written doesn't reproduce.
I have also rewritten the example using a simulated nidcpower.Session (using a 4190), and that DOES reproduce (so I think that nidmm must be at fault).

I have found that this is very particular, and if I modify the code slightly I can stop seeing the error (move the dmm.initiate() after opening the niswitch.Session and its fine).

### System report

`python -c "import nidmm; nidmm.print_diagnostic_information()"` output:

```
OS:
    Name:      Linux
    Version:   #26~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Jul 13 16:27:29 UTC 2
    Bits:      64
Driver:
    Name:      NI-DMM
    Version:   Unknown
Module:
    Name:      nidmm
    Version:   1.4.5
Python:
    Version:   3.9.15 (main, Aug 11 2023, 09:46:20)
[GCC 11.4.0]
    Bits:      64
    Is_Venv:   False
    Installed Packages:
        virtualenv==20.24.2
        setuptools==58.1.0
        platformdirs==3.10.0
        pip==22.0.4
        niswitch==1.4.5
        nidmm==1.4.5
        nidcpower==1.4.5
        hightime==0.2.1
        filelock==3.12.2
        distlib==0.3.7
```

### Steps to reproduce issue

``` python
import niswitch
import nidmm

with nidmm.Session(resource_name="", options= { "simulate": True, "driver_setup": { "Model": "4071" } }) as dmm_session:
    dmm_session.initiate()
    with niswitch.Session(resource_name="", topology="2535/1-Wire 4x136 Matrix", simulate=True) as session1:
        pass
```

Run the above code using python 3.9.15 on Linux

ni-jfitzger, 2023-08-25:
@mgallatin @marcoskirsch
Well, this is a bit embarrassing for us. It looks like I created a similar issue 3 years ago and we still haven't addressed it.
#1480

marcoskirsch, 2023-08-25:
Seems like this issue happens with the older 4070/4071/4072 DMMs but not their modern replacements PXIe-4080/4081/4082. I expect the newer boards to be more common out in the field especially in newer applications.

Unknown at the moment if the problem happens with any of the 4065 DMMs.

We don't know root cause for this, and we don't know if the driver runtime is doing something bad. But we do know that Python 3.10 doesn't exhibit the issue and this is probably a fairly reasonable workaround for anyone running into this out in the field (we'd still want to know, of course).