ni/nimi-python #2007: niswitch.errors.DriverError: -1074118644: No legacy device supports the specified topology

manishkumaretn, 2023-09-06:
Hi,
I am getting following error while trying to open niswitch session in python.
niswitch.errors.DriverError: -1074118644: No legacy device supports the specified topology. Use a different topology, or configure the device under DAQmx and use the DAQmx resource descriptor.

Python code:
import niswitch
resource = "visa://192.168.1.1/PXI8::8::INSTR"
topology1 = "2566/16-SPDT"

with niswitch.Session(resource, topology1) as session:
session.connect(channel1='k0', channel2='com0')

Test environment details:
3.6.0 (v3.6.0:41df79263a11, Dec 23 2016, 07:18:10) [MSC v.1900 32 bit (Intel)]
Package Version

future 0.18.3
hightime 0.2.1
nidaqmx 0.5.7
nifpga 22.0.0
niswitch 1.4.5
numpy 1.19.5
pip 21.3.1
setuptools 28.8.0
six 1.16.0
[image]

please let me know if any additional information is required.

Thanks,
Manish

tobiasgordon, 2023-09-06:
NI-SWITCH does not support VISA resource descriptors. You should use the resource name (from your screenshot, "PXI2Slot18").
This is what the error message is trying to tell you, but it could admittedly be clearer.

I'm going to close this issue. Feel free to reply if you have further questions.

closed 2023-09-06 by tobiasgordon

manishkumaretn, 2023-09-07:
I am getting following error if use "PXI2Slot18" as resource name:
    raise DriverError(code, description)
niswitch.errors.DriverError: -1074118654: Invalid resource name.

Invalid Identifier: PXI2Slot18

tobiasgordon, 2023-09-07:
Based on your screenshot of MAX, I would expect that resource name to work.
Do you have NI-SWITCH installed? (The driver, not the Python wrapper.) I'm assuming yes or else I would expect a different error about a missing library. Which version?
Are you able to run Reset or Self Test on the board? Does the "Test Panels" button work correctly from MAX? Are you able to access the board using a different programming language?

As an aside, note that the version of the ni-switch Python library you are using does not support Python 3.6. See the comment on https://pypi.org/project/niswitch/ (you want 3.7 or newer).

manishkumaretn, 2023-09-11:
Hi,
I installed python 3.7. But i am still getting the same error. 
NI-SWITCH (version -18) is installed.  I can also run the example labview code for PXI-2566 relay module and its working.
[image]
[image]

I can "Reset" and "Self Test" on board using MAX. sharing the screenshot for your reference.
[image]
[image]

"Test  Panels" is not working. Getting following error:
[image]

I have not worked with different programming language.
Thanks,
Manish

tobiasgordon, 2023-09-11:
Manish,

I apologize, I overlooked the fact that you are using a remote system before. The NI-SWITCH API does _not_ directly support accessing devices on remote systems. That is why you are receiving a "resource not found" error - the driver is only looking at the local system.

You have a number of options here (if you want/need to use Python), including:

1. Use LabVIEW Real-Time Module and the LabVIEW Python nodes (e.g. run it locally via LabVIEW). This is the only method officially provided by NI today.
2. [gRPC](https://knowledge.ni.com/KnowledgeArticleDetails?id=kA03q000000oxQGCAY&l=en-US)
3. [rpyc](https://rpyc.readthedocs.io/en/latest/)
4. Run the code locally on the target (via ssh or some other tool)

marcoskirsch, 2023-09-11:
Your remote system is NI Linux RT, right?
You should SSH to it and run the code locally.

manishkumaretn, 2023-09-12:
My system is configured as Pharlap RT.

manishkumaretn, 2023-09-12:
If SSH is applicable for RT Pharlap, can you pl share some more details about usage of SSH.

manishkumaretn, 2023-09-12:
[image]