ni/nimi-python #2024: Incorrect documentation for the channel_name parameter of niFgen.

gfisher-NI, 2023-09-25:
### Description of issue

In the documentation for niFgen.Session the parameter channel_name is listed as having a default value of 'None':
[image]

Further down in the documentation, channel_name is listed as having a default value of '0':
[image]

My understanding is that these are two different things -- on a two-channel device, a default value of '0' would indicate a session contained only channel 0, whereas a value of None would indicate it contained all channels (same as leaving the terminal unwired or wired to a blank string in LabVIEW).

ni-jfitzger, 2023-10-02:
@gfisher-NI thank you for filing a public issue so that users can be aware of this mistake.

If you haven't already, I recommend you also file an internal bug. This description was taken from the niFgen_InitializeWithIndependentChannels documentation in the C Reference and it doesn't appear to have changed.

gfisher-NI, 2023-10-02:
@ni-jfitzger  Will do! I just know the preference is to open issues involving Python here, first.