ni/nimi-python #2036: nidcpower.errors.DriverError: -1074097808: Could not load the dynamic library.

kwangho9, 2024-01-04:
### Description of issue

In PXIe-4162, Session initialization works well with simulation, but in PXIe-4139, an error occurs.

### System report

`python -c "import session; session.pxie4139()"` output

    D:\nidcpower>python -c "import session; session.pxie4139()"
    PXIe-4139 Simulate Session init()
    Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "D:\nidcpower\session.py", line 13, in pxie4139
      example1(resource, options)
    File "D:\nidcpower\session.py", line 6, in example1
      with nidcpower.Session(resource_name=resource, options=options) as session:
    File "C:\Users\khjung\AppData\Local\Programs\Python\Python310\lib\site-packages\nidcpower\session.py", line 7244, in __init__
      self._interpreter.set_session_handle(self._fancy_initialize(resource_name, channels, reset, options, independent_channels))
    File "C:\Users\khjung\AppData\Local\Programs\Python\Python310\lib\site-packages\nidcpower\session.py", line 7557, in 
    _fancy_initialize
      return self._initialize_with_independent_channels(resource_string, reset, option_string)
    File "C:\Users\khjung\AppData\Local\Programs\Python\Python310\lib\site-packages\nidcpower\session.py", line 7984, in 
    _initialize_with_independent_channels
    vi = self._interpreter.initialize_with_independent_channels(resource_name, reset, option_string)
     File "C:\Users\khjung\AppData\Local\Programs\Python\Python310\lib\site-packages\nidcpower\_library_interpreter.py", line 494, in initialize_with_independent_channels
      errors.handle_error(self, error_code, ignore_warnings=False, is_error_handling=False)
    File "C:\Users\khjung\AppData\Local\Programs\Python\Python310\lib\site-packages\nidcpower\errors.py", line 123, in handle_error
      raise DriverError(code, description)
     nidcpower.errors.DriverError: -1074097808: Could not load the dynamic library.

     Device: PXI1Slot1
     Library Name: nidcpowerPAL.dll

    The parameter received by the function is not valid.

     D:\nidcpower>

`python -c "import session; session.pxie4162()"` output

    D:\nidcpower>python -c "import session; session.pxie4162()"
    PXIe-4162 Simulate Session init()
    Session Initialize Done.

    D:\nidcpower>

### Steps to reproduce issue

**session.py**
```
import nidcpower

resource = 'PXI1Slot1'
def example1(resource, options):
    with nidcpower.Session(resource_name=resource, options=options) as session:
        print('Session Initialize Done.')

def pxie4139():
    print("PXIe-4139 Simulate Session init()")
    options = {'simulate':True, 'driver_setup':{'Model':'4139', 'BoardType':'PXIe', }, }
    example1(resource, options)

def pxie4162():
    print("PXIe-4162 Simulate Session init()")
    options = {'simulate':True, 'driver_setup':{'Model':'4162', 'BoardType':'PXIe', }, }
    example1(resource, options)

if __name__ == '__main__':
    pxie4139()
```

1. `python -c "import session; session.pxie4139()" run -> error.
2. `python -c "import session; session.pxie4162()" run -> pass.
3. other model("4130", "4135", "4138" ...) all error.

tobiasgordon, 2024-01-04:
The PXIe-4162 uses a different driver stack than the other devices you listed. Do you have the PAL-based instrument support installed?
Please check that first. If that doesn't work, we need more information: OS, NI-DCPower version, etc.

marcoskirsch, 2024-01-05:
There's nothing indicating a bug in the `nidcpower` Python module. Rather this appears to be an installation issue on your system. We can provide some help, but you need to provide a lot more information: what OS are you using, what Python version and bitness, etc.

closed 2024-01-05 by marcoskirsch