ni/nimi-python #2039: Github Actions Workflows should be updated to use Node.js 20

ni-jfitzger, 2024-02-09:
### Description of issue

Our nimibot [runner summaries are showing annotations](https://github.com/ni/nimi-python/actions/runs/7836439507):

>[system_test (niscope)](https://github.com/ni/nimi-python/actions/runs/7836439507/job/21384097974)
Node.js 16 actions are deprecated. Please update the following actions to use Node.js 20: actions/checkout@v3, codecov/codecov-action@v3. For more information see: https://github.blog/changelog/2023-09-22-github-actions-transitioning-from-node-16-to-node-20/.

We should update our configuration to use Node.js 20, before support for Node.js 16 is dropped entirely.

The configuration for this can be found in `.github\workflows\github_actions_aws_*.yml` (the * is a wildcard not part of a filename).
The steps look like:
```
    steps:
      - name: checkout repository
        uses: actions/checkout@v3
      - name: upload coverage
        uses: codecov/codecov-action@v3
```

* For codecov-action, [the issue was addressed in the v4 release](https://github.com/codecov/codecov-action/issues/1226)
  * Arguments to the action have changes, so see the [README](https://github.com/codecov/codecov-action/blob/main/README.md)
  * There's some mixed messaging that suggests we could go back to Tokenless uploading (PRs made from forks to the upstream public repos will support tokenless), but based on my reading of things, I don't think that's the case. It'll give us intermittent upload failures, again
* For checkout, [the issue was addressed in the v4 release](https://github.com/actions/checkout/releases/tag/v4.0.0)
  * For usage, see the [README](https://github.com/actions/checkout/blob/main/README.md)

bkeryan, 2024-02-09:
@ni-jfitzger Consider using the Mend Renovate app to keep your GitHub actions and Poetry lock files up-to-date. Here is the config we've been using in measurementlink-python: https://github.com/ni/measurementlink-python/blob/main/.github/renovate.json

It automatically posts PRs for dependency updates and you can configure scheduling and auto-merging if you want. It also creates a "Dependency Dashboard" issue that you can use to see its status and control it: https://www.github.com/ni/measurementlink-python/issues/475

ni-jfitzger, 2024-02-09:
@bkeryan That sounds very useful. I'll create an issue under our Streamlined Processes milestone.

ni-jfitzger, 2026-08-25:
Now, it's probably Node.js 24.