ni/nimi-python #2061: nidcpower.errors.DriverError: -1074135040: IVI: Unrecoverable Failure.

chesterknurek, 2024-05-09:
import nidcpower
# Configure the session.

with nidcpower.Session(resource_name='PXI1Slot3/0') as session:

This snippet of code produces this error: nidcpower.errors.DriverError: -1074135040: IVI: Unrecoverable Failure.

I have qty 2 PXIe-4163 sitting inside a PXIe-1084 mainframe with PXIe-8301 Thunderbolt interface card.

I reset the configuration data in MAX and tried renaming the cards but no luck.

marcoskirsch, 2024-05-10:
Hello

You have a legitimate problem in your system, but it is not a bug in the `nidcpower` Python module. The Python module is loading the driver runtime and calling into it correctly and the driver runtime is returning this error.

The more appropriate forum for this sort of issue is [NI Discussion Forums](https://forums.ni.com/t5/Discussion-Forums/ct-p/discussion-forums) or to contact NI support directly.

We have seen lots of problems with buggy Thunderbolt implementation in several vendor PCs. When you reach out to support be sure to include enough information where they can help you:

* PC this running on
* Operating System
* NI-DCPower driver runtime version
* Screenshot of what you see in MAX when you select the board (the screen where calibration data is shows)
* Can you reset in MAX?
* Can you self test in MAX?
* Have you used this NI hardware in any other configurations successfully?

FYI @tobiasgordon 

Closing the issue.

closed 2024-05-10 by marcoskirsch

chesterknurek, 2024-05-10:
Thankyou for the immediate reply.

I will answer the items below (for your records).

For the record, I have qty 2 systems, one having Thunderbolt and the other having MXI interfaces.
Both (MXI and thunderbolt) have issue so it is not entirely Thunderbolt problem.

-Chet

Del Precision 3430 (i7)

Windows 10 Corporate

YES

YES

YES, using C# language, qty 3 PXIe-4163s work flawlessly, together, in a homebrew silicon photonics test station here at Cisco.

marcoskirsch, 2024-05-10:
> YES, using C# language, qty 3 PXIe-4163s work flawlessly, together, in a homebrew silicon photonics test station here at Cisco.

Can you run a C# program  that uses NI-DCPower .NET API on the same system in which you see the Python failures? That would be very strange!

chesterknurek, 2024-05-10:
The C# app is compiled with:

NationalInstruments.Common
NationalInstruments.ModularInstruments.Common
NationalInstruments.ModularInstruments.ModualInstrumentsSystem
NationalInstruments.ModularInstruments.NIDCPower.Fx45

I did notice the error that the Python driver reported indicated an IVI: Unrecoverable Failure.

I don’t know IVI, but there is a separate tab in MAX for a configuration one can setup using NIDCPOWER.
Is IVI required?

I have qty 3 SMUS here in NJ. And qty SMUS in CA.

They are $33K each, so maybe I am the first trying the python driver….

Thankyou for your replies.

Whatever I can do on this end to assist, let me know.

-Chet

chesterknurek, 2024-05-10:
I am using the .NET Framework 4.6.1

chesterknurek, 2024-05-10:
NIDCPower 21.5.0

C:\Program Files (x86)\IVI Foundation\IVI\Microsoft.Net\Framework32\v4.5.50709\NationalInstruments.ModularInstruments.NIDCPower 21.5.0\NationalInstruments.ModularInstruments.NIDCPower.Fx45.dll

marcoskirsch, 2024-05-10:
> I did notice the error that the Python driver reported indicated an IVI: Unrecoverable Failure.
> I don’t know IVI, but there is a separate tab in MAX for a configuration one can setup using NIDCPOWER.
Is IVI required?

NI-DCPower uses IVI under the hood for some things, and the error code is either coming from IVI or is coming from the driver but using an error code defined by IVI.

> I have qty 3 SMUS here in NJ. And qty SMUS in CA.
> They are $33K each, so maybe I am the first trying the python driver….

You are most definitely not the first one using PXIe-4163 with out Python module.

> Whatever I can do on this end to assist, let me know.

What I asked before: Can you run a C# program that uses NI-DCPower .NET API **on the same system** in which you see the Python failures?

chesterknurek, 2024-05-10:
Yes.
I first tried the python driver with the  Thunderbolt config (NJ) and then with  the MXI config (CA).
Same error on both systems.

marcoskirsch, 2024-05-10:
> Yes.

Ok, so a problem with the `nidcpower` Python module is completely ruled out. I think you should go through NI support, Please provide as much detail as you can including troubleshooting steps. Good luck!

chesterknurek, 2024-05-10:
I coded a small command line app with arguments to config a channel on the SMU in C#.
Works.

-Chet

chesterknurek, 2024-05-10:
Thankyou Marcos,

-Chet