ni/nimi-python #2063: Remotely connecting to a PXIe-7139 card located at some IP address

heydaridlm, 2024-07-17:
### Description of issue

I have a PXIe-1075 chassis with some PXIe-4139 cards.  I have connected the PXIe-1075 to my PC using a crossover ethernet cable and established an intranet between that PC and the embedded PC on the 1075.  
I tried to use VISA to communicate with the PXIe card (located at, e.g. `visa://xxx.xxx.x.x/PXIyy::0::INSTR`).  'x' and 'y' are numbers.

     import nidcpower
     addr = 'visa://xxx.xxx.x.x/PXIyy::0::INSTR'
     session =  nidcpower.Session(resource_name=addr)

This results in the following error

     InvalidRepeatedCapabilityError: An invalid character (Multiple '-' or ':') was found in repeated capability string

Perhaps `nidcpower` does not support connecting to remote PXI cards?

### Steps

1. Establish an intranet between external computer and PXI chassis
2. Verify PXI resource is accessible by external computer through `pyvisa` (even though PXI instrument does not support VISA commands)
3. Attempt to open that same resource with `nidcpower`: FAIL

tobiasgordon, 2024-07-18:
You are correct, NI-DCPower (and other drivers supported by nimi-python) do not support remote access via VISA.

That said, you have options!

NI gRPC is one option. See https://knowledge.ni.com/KnowledgeArticleDetails?id=kA03q000000oxQGCAY&l=en-US for more details. 

Others include using other remote calls. We have had success using rpyc as well, which is quite lightweight.

ni-jfitzger, 2025-01-26:
@marcoskirsch @tobiasgordon 
This issue has been tagged invalid (because it's a question about NI Drivers and not related our Python APIs. Therefore, I'm closing it.

closed 2025-01-26 by ni-jfitzger